embassy-rs/embassy #919: WASM example cannot import `env`

sjmackenzie, 2022-08-22:
Steps to repro:

* checkout 1b9599025868d3a5d0d8e773593b05df8b2fecf2
* run `[/path/to/embassy/examples/wasm]$ wasm-pack build --target web`
* run `[/path/to/embassy/examples/wasm]$ simple-http-server`
* navigate to http://0.0.0.0:8000 with your browser
* click on `index.html`

Expected behaviour:
See log entries printed via an html list

Actual behaviour:
* Blank white screen with the word "log" printed at the top but nothing else. 
* This error is printed in the console: `Uncaught TypeError: Error resolving module specifier “env”. Relative module specifiers must start with “./”, “../” or “/”.`

lulf, 2022-08-23:
This is related to critical-section no longer being 'importable'. I'll dig and see if I can fix it.

lulf, 2022-08-23:
Fixed in #925 and works for me now, please reopen if you still face some issues with it.

sjmackenzie, 2022-08-24:
The `actual behaviour` (as above) persists in 529535194d4b5d58b31fd6a7541176105e3c63f7

Changes to steps: I updated my dependencies as per https://github.com/embassy-rs/embassy/commit/bd4ae2e9526cb66c15b7335cfbc803106637c7e1 (and I'm using `wasm-pack --version 0.9.1`)

Perchance did you clear your artefacts before testing?

```

lulf, 2022-08-24:
@sjmackenzie 

What

lulf, 2022-08-24:
@sjmackenzie I started from a clean repo (so no existing artifacts), and that works for me now. Are you modifying the example with some additional dependencies? Basically this error is "unresolved symbol" error, so if a dependency is not "wasm-ready", you can get this error.

Could you share your .wasm file that is generated in `pkg/embassy_wasm_example_bg.wasm`?

sjmackenzie, 2022-08-24:
Oh, I'm being an idiot, a browser hard refresh solved the issue. It's working!

Thank you for you patch and your patience.

closed 2022-08-24 by sjmackenzie