embassy-rs/embassy #953: rp2040 async SPI read never wakes

newAM, 2022-09-18:
Async SPI reads on the rp2040 never wake.  As far as I have debugged this occurs because the DMA interrupt handler is never called.

I modified the [spi_async](https://github.com/embassy-rs/embassy/blob/336ebe54c0f32fd729d50889850032222c6c6ca4/examples/rp/src/bin/spi_async.rs) example for a minimal reproduction:

```rs
#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]

use defmt::*;
use embassy_executor::Spawner;
use embassy_rp::spi::{Config, Spi};
use embassy_time::{Duration, Timer};
use {defmt_rtt as _, panic_probe as _};

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let p = embassy_rp::init(Default::default());
    info!("Hello World!");

    let miso = p.PIN_12;
    let mosi = p.PIN_11;
    let clk = p.PIN_10;

    let mut spi = Spi::new(p.SPI1, clk, mosi, miso, p.DMA_CH0, p.DMA_CH1, Config::default());

    loop {
        let mut rx_buf: [u8; 6] = [0; 6];
        spi.read(&mut rx_buf).await.unwrap();
        info!("{:?}", rx_buf);
        Timer::after(Duration::from_secs(1)).await;
    }
}
```

newAM, 2022-09-18:
Looked into the code, the RX buffer will never fill because the hardware requires a TX to clock data into the receive FIFO.

I changed the read function to this and it works as I expected, but now I am questioning if my expectations of the `read` function are correct.

```rs
    pub async fn read(&mut self, buffer: &mut [u8]) -> Result<(), Error> {
        self.transfer_inner(buffer, buffer).await
    }
```

newAM, 2022-09-18:
The behavior of the RP2040 seems dissimilar to the STM32 which calls `write_repeated` to clock the bus during a `read`:
https://github.com/embassy-rs/embassy/blob/336ebe54c0f32fd729d50889850032222c6c6ca4/embassy-stm32/src/spi/mod.rs#L477

newAM, 2022-09-18:
On a related topic, the RP2040 `write` does not clear the receive FIFO, resulting in stale data when a write is followed by a transfer.

newAM, 2022-09-18:
I have my version of a fix here: https://github.com/newAM/embassy/commit/93476cd0e23ac2833408bc5b94c9dd958706c7ae

if this is a bug I can open a PR with that.

Dirbaio, 2022-09-18:
Implementing `read` with `transfer` feels a bit dangerous, because it "leaks" the previous buffer contents on the MOSI pin.

Instead you could do a DMA write from a buffer with a single `0x00` byte, without memory address autoincrement so that it tx's the 0x00 byte N times.

newAM, 2022-09-18:
> Instead you could do a DMA write from a buffer with a single `0x00` byte, without memory address autoincrement so that it tx's the 0x00 byte N times.

Done, opened a PR at #954

commit 66633902 (2022-09-19): Merge #954

954: rp: fix async SPI read and write r=lulf a=newAM

Closes #953
PR #954 (2022-09-19): rp: fix async SPI read and write
Closes #953
closed 2022-09-19 by bors[bot]