embassy-rs/embassy #1001: Impliment Embedded-hal PWM for stm32 simple_pwm

andyblarblar, 2022-10-07:
It would be very useful if [simple pwm](https://docs.embassy.dev/embassy-stm32/git/stm32f767zi/pwm/simple_pwm/struct.SimplePwm.html) implimented [embedded hals pwm trait](https://docs.rs/embedded-hal/latest/embedded_hal/trait.Pwm.html), allowing for easier use with other periferal driver crates.

It looks like the current implimentation contains most of what we would need to do so, but I'm not sure if there's anything interally preventing this.

andyblarblar, 2022-10-07:
> The trait is removed in embedded-hal 1.0, I'm not sure if we want to encourage its use.
> 
> What driver are you trying to use that needs the trait?

I'm planning on writing a driver for the little LD06 LiDAR, which requires some sort of motor control over PWM. I'd like to make it generic across both embassy and RTIC using the stm32f7xx hal, which is why this trait would be useful. Even if embedded hal removes that trait in 1.0, I'm not sure if that changes much, since there doesn't seem to be any sort of replacemet.

leftger, 2026-10-09:
This looks like it was addressed — the embedded-hal PWM traits are now implemented.

`embassy-stm32/src/timer/simple_pwm.rs` has `impl embedded_hal_02::Pwm for SimplePwm` (line ~588) and `impl embedded_hal_1::pwm::SetDutyCycle for SimplePwmChannel` (line ~557).

Reopen or comment if you needed a trait version or method that still isn't covered.

closed 2026-10-09 by leftger