embassy-rs/embassy #1007: embassy-boot change memory layout STM32WL

miathedev, 2022-10-10:
Hi,

i tryed to use the complete memory of the STM32WL for embassy boot (example provided: https://github.com/embassy-rs/embassy/tree/master/examples/boot).
To try things out i changed memory.x from:
```
MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  FLASH                             : ORIGIN = 0x08000000, LENGTH = 24K
  BOOTLOADER_STATE                  : ORIGIN = 0x08006000, LENGTH = 4K
  ACTIVE                            : ORIGIN = 0x08008000, LENGTH = 32K
  DFU                               : ORIGIN = 0x08010000, LENGTH = 36K
  RAM                         (rwx) : ORIGIN = 0x20000000, LENGTH = 16K
}

__bootloader_state_start = ORIGIN(BOOTLOADER_STATE) - ORIGIN(FLASH);
__bootloader_state_end = ORIGIN(BOOTLOADER_STATE) + LENGTH(BOOTLOADER_STATE) - ORIGIN(FLASH);

__bootloader_active_start = ORIGIN(ACTIVE) - ORIGIN(FLASH);
__bootloader_active_end = ORIGIN(ACTIVE) + LENGTH(ACTIVE) - ORIGIN(FLASH);

__bootloader_dfu_start = ORIGIN(DFU) - ORIGIN(FLASH);
__bootloader_dfu_end = ORIGIN(DFU) + LENGTH(DFU) - ORIGIN(FLASH);
```

to:

```
MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  FLASH                             : ORIGIN = 0x08000000, LENGTH = 24K
  BOOTLOADER_STATE                  : ORIGIN = 0x08006000, LENGTH = 4K /* 2x4k?*/
  ACTIVE                            : ORIGIN = 0x08008000, LENGTH = 40K
  DFU                               : ORIGIN = 0x08012000, LENGTH = 40K
  RAM                         (rwx) : ORIGIN = 0x20000000, LENGTH = 16K
}

__bootloader_state_start = ORIGIN(BOOTLOADER_STATE) - ORIGIN(FLASH);
__bootloader_state_end = ORIGIN(BOOTLOADER_STATE) + LENGTH(BOOTLOADER_STATE) - ORIGIN(FLASH);

__bootloader_active_start = ORIGIN(ACTIVE) - ORIGIN(FLASH);
__bootloader_active_end = ORIGIN(ACTIVE) + LENGTH(ACTIVE) - ORIGIN(FLASH);

__bootloader_dfu_start = ORIGIN(DFU) - ORIGIN(FLASH);
__bootloader_dfu_end = ORIGIN(DFU) + LENGTH(DFU) - ORIGIN(FLASH);
```

Since the STM32WL has 256KB of Flash, it looked like it should work.

After the change and reuploading bootloader and binary, it doesnt boot anymore.
Why?

And why is there some free space at the Bootloader State Partition?

I have a JLink Probe near me. I actually tryed to somehow debug the bootloader, but after entering the main loop, debugging goes into the void somehow.

Im fresh into Rust and just got my feet rusty. Hope you dont mind me to ask questions like this.

lulf, 2022-10-10:
I think it's great to ask questions like this here, so don't hesitate to do that!

So from looking at your changes, the ACTIVE and DFU partitions are both set to 40kB. The way the bootloader works is that it requires the DFU partition to be 1 page bigger than the ACTIVE partition, as it uses that during the swap process.  See https://github.com/embassy-rs/embassy/blob/master/embassy-boot/boot/src/lib.rs#L123-L146 for details. So in your case, with an active partition of 40 kB, the DFU partition must be 42kB.

I had a look at https://embassy.dev/book/dev/bootloader.html which seems to miss this note, so I think that should be added. Thanks for finding that issue! 

Regarding the space after the state partition, there are no particular reason for that. For some boards with more flash, the state partition might need to expand, so it would mean less maintenance of the examples to get it working on multiple boards I think.

lulf, 2022-10-10:
FWIW, it's probably worth mentioning this as well https://github.com/embassy-rs/embassy/blob/master/embassy-boot/boot/src/lib.rs#L408  - these are runtime assertions that are likely the cause of why the bootloader is not progressing.

Ideally we'd catch that at compile time, but not sure how other than linker script assertions.

miathedev, 2022-10-11:
Thanks a lot! Its working now!

I just put here my working memory.x for people to look up.
```
MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  BOOTLOADER                        : ORIGIN = 0x08000000, LENGTH = 24K
  BOOTLOADER_STATE                  : ORIGIN = 0x08006000, LENGTH = 4K
  FLASH                             : ORIGIN = 0x08008000, LENGTH = 112K
  DFU                               : ORIGIN = 0x08024000, LENGTH = 114K
  RAM                         (rwx) : ORIGIN = 0x20000000, LENGTH = 32K
}

__bootloader_state_start = ORIGIN(BOOTLOADER_STATE) - ORIGIN(BOOTLOADER);
__bootloader_state_end = ORIGIN(BOOTLOADER_STATE) + LENGTH(BOOTLOADER_STATE) - ORIGIN(BOOTLOADER);

__bootloader_dfu_start = ORIGIN(DFU) - ORIGIN(BOOTLOADER);
__bootloader_dfu_end = ORIGIN(DFU) + LENGTH(DFU) - ORIGIN(BOOTLOADER);
```

closed 2022-10-11 by miathedev