embassy-rs/embassy #1013: STM32F0 example failed to build for dev profile

linfan68, 2022-10-18:
Basically, running `cargo build` inside `embassy/examples/stm32f0/` failed on current master:
[18453ee64c7d1b397aa96cb322cedee6166b4602]
With follow error:
```
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "/tmp/rustcdKK4gK/symbols.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.17dzbbc7j4yq3dif.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.1smhw4b7s26ptds8.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.20pb365vzfg1v2av.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.2dg3v2dyinp02jis.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.2umlfi37mesypxl1.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.2y8xrqgxb9au64gw.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.34nnhg3eokttlkch.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.3914hv16yx7cpybu.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.3b5f7wr66c6rucc9.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.3bgpl9lhv826pe1r.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.3d4017ybf9nvwult.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.3dj6kcygit36294o.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.3eq0qsh4xe0hn58a.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.3pajjy9ydr7f8mnr.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.4552689287fiymn6.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.49kfx0rhgfpxm669.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.4a40klnwbph00lvh.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.4fxvnxx7fn5bbhtj.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.4mctq3qjl5yexk6.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.5etjht57f74yy96.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.e7rte67qid4r6xt.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.ff5ckxv2lglhmb1.rcgu.o" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/hello-3cd841e7f3de83d0.v1kg5ogtdjsgoou.rcgu.o" "--as-needed" "-L" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps" "-L" "/workspaces/embassy/examples/stm32f0/target/debug/deps" "-L" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/build/cortex-m-b4373e67c2a50c23/out" "-L" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/build/cortex-m-rt-eed8c942d4bade10/out" "-L" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/build/defmt-365addd4d9d21ab4/out" "-L" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/build/stm32-metapac-e09df70eb1ce83cb/out/src/chips/stm32f031f6" "-L" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/build/stm32-metapac-e09df70eb1ce83cb/out/src/chips/stm32f031f6/memory_x/" "-L" "/usr/local/rustup/toolchains/nightly-2022-09-22-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib" "-Bstatic" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libcortex_m_rt-9ee1337b6f89b9b5.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libpanic_probe-fbe2e21b3edfa6b7.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libdefmt_rtt-95a21cdd7f5059ee.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libembassy_stm32-c9240fe5b0c75a56.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libembedded_io-8b3d36242042bd2d.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libembassy_embedded_hal-6fda8c0951f694e0.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libembedded_hal_async-1b46fd30e3576229.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libembedded_hal-17873bda88e2a26b.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libembedded_storage_async-4a77143f3cf95c28.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libembedded_storage-a70b925128799502.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libstm32_metapac-f4ced5350063eeb4.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libembassy_sync-3beca3085d0ab8ce.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libheapless-250c185dbe3efc1c.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libstable_deref_trait-66f97325b71861ae.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libatomic_polyfill-db988762ae3d4e48.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libcritical_section-1d13f9afe8ec6d7b.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libbare_metal-c116ef0977dfe2ec.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libhash32-530d4dbae19cacdb.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libbyteorder-ec637884220478e1.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libembassy_cortex_m-82485f9fe4396d2e.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libembassy_hal_common-bf5ab1e94505f663.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libnum_traits-1bca52d0beaf51ac.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libcortex_m-45c4ce200b689b80.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libvolatile_register-8b130a45c77c6e0e.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libvcell-61e48036dfb7bd31.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libbare_metal-5d2a91cf452c7c68.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libembassy_executor-81462aced98d0a22.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libcfg_if-868cc9c10f3c052c.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libstatic_cell-d11754f28dad21b5.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libembassy_time-e2446ed3e29029c3.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libfutures_util-bb805c2884b266fa.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libpin_project_lite-09ebd436cafe2639.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libfutures_sink-3895e8df72ddd570.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libfutures_task-465546750570354f.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libpin_utils-f38cf52c7bbf39c6.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libfutures_core-9ffc7c05c54e1962.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libembedded_hal-2bb67af7af8af60d.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libvoid-8d4c8d84b55a6569.rlib" "/workspaces/embassy/examples/stm32f0/target/thumbv6m-none-eabi/debug/deps/libnb-d38719cce06efad1.rlib" "/workspaces/embassy/examples/stm32f0 [... cut ...]
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 1010 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 1024 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 1140 bytes
          rust-lld: error: too many errors emitted, stopping now (use --error-limit=0 to see all errors)
```

It seems like no enough FLASH for the program, but the warning messages seems fishy.
But release build `cargo build --release` seems fine, and program can run on device ok.

Any thoughts?

Dirbaio, 2022-10-18:
The issue is there's not enough flash, yep.

linfan68, 2022-10-19:
@Dirbaio  How about this warning?
```
 rust-lld: warning: section type mismatch for .got
          >>> <internal>:(.got): SHT_PROGBITS
          >>> output section .got: SHT_NOBITS
```
seems like some linker script issue to me, what's the story behind this?
(Sorry I'm pretty comfortable with embedded systems, but rather new to Rust.)

Dirbaio, 2022-10-26:
I don't know what it is, but it's ignorable. It shows up in pretty much any rust embedded project, but you can only see it when there's an actual linker error since Cargo doesn't print the linker output otherwise.

closed 2022-10-26 by Dirbaio