embassy-rs/embassy #1035: STM32WL RTC Peripheral Question

miathedev, 2022-10-29:
**Dear embassy Team,**

i got my feet rusty and found the stm32 **time_driver**.
Is it true, that embassys "time_driver" is actually not a "real" RTC Driver, its using a Timer for time keeping and not the actual RTC? And its actually only something like systick, but self-made and more accurate?

**_Theory question_**
In my university we didn't talk a lot about RCC. Did i read the Source Code and Datasheet right and the RCC pipes a divided System-Clock to the Timer Peripheral programmed in the "time_driver", which is then, used as a since boot relative time source?

**_subghz_**
Did i read the datasheet correct, and the Radio Modem has its own RTC?

**_Why_**

Im asking, because i would like to actually use the RTC for "real time" time keeping synced with a gps. If i understood everything correct, the RTC is not used by embassy at all?

Sorry for bothering. Im quite new into Rust and actually first time i am diving deep into micro-controllers by my self.

lulf, 2022-10-30:
> **Dear embassy Team,**
> 
> i got my feet rusty and found the stm32 **time_driver**. Is it true, that embassys "time_driver" is actually not a "real" RTC Driver, its using a Timer for time keeping and not the actual RTC? And its actually only something like systick, but self-made and more accurate?
> 

You are correct, it only supports using the timer peripherals at the moment. You can configure the timer resolution using cargo feature flags for the embassy-time crate. More on timer [here](https://embassy.dev/book/dev/runtime.html#_time). 

I think using the RTC peripheral is beneficial if you have the need to wait for very long times before waking up the MCU and the additional low power you get from that, but it is not implemented as a possible time driver yet.

> **_Theory question_** In my university we didn't talk a lot about RCC. Did i read the Source Code and Datasheet right and the RCC pipes a divided System-Clock to the Timer Peripheral programmed in the "time_driver", which is then, used as a since boot relative time source?
> 

I think that makes sense. I would say RCC is a 'special' peripheral for setting up the 'clock tree' of the MCU. It can configure the clock source, either from the CPU, an internal clock, or from an external crystal usually. 

The time driver is independently of that configured with a 'tick rate' at compile time (if you enable it), and it will configure alarms to wake up at the correct time no matter which clock source you configured in RCC (within reasonable limits, you can't tick faster than the clock etc.).

> **_subghz_** Did i read the datasheet correct, and the Radio Modem has its own RTC?
> 

Not sure exactly which STM32WL you are referring to, but for the STM32WL55, the SUBGHZ peripheral can use different clock sources: two different internal RC or a high precision external clock, but it's not 'dedicated' to the SUBGHZ peripheral. But as far as I understand, there is no RTC for the radio modem, just clock sources used in different radio modes.

> **_Why_**
> 
> Im asking, because i would like to actually use the RTC for "real time" time keeping synced with a gps. If i understood everything correct, the RTC is not used by embassy at all?
> 

For the time being, you would have to compile embassy-stm32 with the 'unstable-pac' feature to expose the RTC registers and configure it 'manually' using the registers, and/or submit a PR to implement a HAL interface for the RTC and potentially use it as a time driver too :) 

> Sorry for bothering. Im quite new into Rust and actually first time i am diving deep into micro-controllers by my self.

Asking questions this is totally fine, I hope you got some helpful information!

snorkman88, 2023-07-22:
> I think using the RTC peripheral is beneficial if you have the need to wait for very long times before waking up the MCU and the additional low power you get from that, but it is not implemented as a possible time driver yet.

@lulf Thanks for clarifying this point. Atm, I am porting a code that depends on the RTC to wake up the STM32F411.

In line with what you mentioned, I was using the RTC as a source of IRQ every 10 minutes and this would allow me to put the microcontroller in sleep mode reducing its consumption to the order of a few microamps and resume operation when the interruption takes place.

Is there a plan to have such feature available in the [rtc mod ](https://github.com/embassy-rs/embassy/blob/main/embassy-stm32/src/rtc/mod.rs) any time soon? Thanks !