embassy-rs/embassy #1041: embassy-lora `Sx126xRadio` pin types

jbeaurivage, 2022-11-03:
I'm currently trying to use `Sx126xRadio` that #985 has brought. As defined [here](https://github.com/embassy-rs/embassy/blob/b99533607ceed225dd12ae73aaa9a0d969a7365e/embassy-lora/src/sx126x/mod.rs#L26), it seems that there is only one type parameter for all output pins, and one type parameter for all interrupt-input pins:

```rust
impl<SPI, CTRL, WAIT, BUS> Sx126xRadio<SPI, CTRL, WAIT, BUS>
where
    SPI: SpiBus<u8, Error = BUS> + 'static,
    CTRL: OutputPin + 'static,
    WAIT: Wait + 'static,
    BUS: Error + Format + 'static,
{
    pub async fn new(
        spi: SPI,
        cs: CTRL,
        reset: CTRL,
        antenna_rx: CTRL,
        antenna_tx: CTRL,
        dio1: WAIT,
        busy: WAIT,
        enable_public_network: bool,
    ) -> Result<Self, RadioError<BUS>> {
        let mut lora = LoRa::new(spi, cs, reset, antenna_rx, antenna_tx, dio1, busy);
        lora.init().await?;
        lora.set_lora_modem(enable_public_network).await?;
        Ok(Self { lora })
    }
}
```

@lulf and @ceekdee, since you've worked on that PR: Unless I'm mistaken, does this mean that all the `CTRL` pins would need to have the same type, and all the `WAIT` pins likewise? Would it make more sense to have a unique parameter for each of these pins?

lulf, 2022-11-03:
Both the nRF and STM32 HALs provide an AnyPin type you can create, i.e. using p.PB4.downgrade(). In general, I think we want to have a RadioSwitch parameter for the antenna pins at least, because it can differ from board to board. I think @ceekdee is planning to fix that.

jbeaurivage, 2022-11-03:
Ah, fair enough. I want to use it with atsamd-hal, but it has a similar DynPin API.

Regarding the antenna pins, I believe my board has a single switch for both TX and RX. How is it recommended to handle that case?

lulf, 2022-11-03:
Until the API has been refactored, I think you maybe need to wrap your pin in some custom shared pin type containing an inner type with a RefCell<AnyPin> that the driver can use. Since TX and RX is not done at the same time, it should be 'safe' to do this.

It's exactly the reason I think we should use a RadioSwitch trait similar to the other drivers, because boards do it so differently.

ceekdee, 2022-11-03:
In the original implementation, each pin did have its own type.  This seemed cumbersome and unnecessary to me, and when I realized that AnyPin could be used for both input and output pins, I changed to use CTRL and WAIT.  This can certainly be re-visited as we attempt to combine MCU/Semtech chip combinations into one implementation (https://github.com/embassy-rs/embassy/issues/1023).  Any additional observations and guidance you wish to provide can be placed in issue 1023.

Regarding the antenna processing, another option until the re-implementation is to make local changes to the following functions in board_specific.rs in the sx126x implementation:

- brd_ant_sleep()
- brd_ant_set_rx()
- brd_ant_set_tx()

to support your use case.

lulf, 2022-11-03:
@ceekdee I might have a go at introducing a RadioSwitch trait for the sx126x in a way that should help this use case, if that's ok with you.

jbeaurivage, 2022-11-03:
Would it make sense to extend the `RadioSwitch` trait to all board-specific functionality, which the user is expected to implement? All of the generic complexity could be hidden behind it so that the radio structs wouldn't have to deal with individual pins. A default struct which implements that trait could be provided, for users that don't have specific needs or can provide type-erased pins.

lulf, 2022-11-03:
By other board-specific functionality, do you mean the Wait, Ready etc. pins? If so, I guess that could be there as well, though I've not seen them having different semantics across boards, vs. the radio pins which seem to vary a lot more.

I'm fine either way, I think it's worth exploring.

ceekdee, 2022-11-03:
Perhaps a better way to envision the "orchestrator" is as a builder which returns a LoRa structure supporting the API for an MCU/LoRa chip combination.

Since the only current interfaces to Semtech chips of which I am aware are SPI and pin setting/checking, the hope is to hide any dependencies on STM32WL or nRF52 for this behind the LoRa API.

lulf, 2022-11-03:
I've moved the discussion over to #1023 now, focusing on this issue here:

> Would it make sense to extend the `RadioSwitch` trait to all board-specific functionality, which the user is expected to implement? All of the generic complexity could be hidden behind it so that the radio structs wouldn't have to deal with individual pins. A default struct which implements that trait could be provided, for users that don't have specific needs or can provide type-erased pins. I can also try to whip up a POC if y'all are interested.

I think that makes sense, maybe have a look at the stm32wl integration in embassy-lora for inspiration and see what you think.

jbeaurivage, 2022-11-07:
@ceekdee, I've gotten started a rough API that abstracts over the physical aspects of a board ([view here](https://github.com/jbeaurivage/embassy/tree/lora-phy)). However while trying it out, I seem to consistently run into a bug in [this function](
https://github.com/embassy-rs/embassy/blob/b99533607ceed225dd12ae73aaa9a0d969a7365e/embassy-lora/src/sx126x/sx126x_lora/mod.rs#L647) where indexing `buffer` overflows when `i` reaches 4. Seems like whe debugging, `number_of_registers` is set to 34, is this normal?

ceekdee, 2023-04-30:
The original sx126x driver has been removed in favor of using the lora-phy crate.  Therefore, closing this issue.

closed 2023-04-30 by ceekdee