embassy-rs/embassy #1048: nrf: use of undeclared crate or module `interrupt`

diondokter, 2022-11-08:
I had the following code that could not compile:

```rust
let irq = embassy_nrf::interrupt::take!(UARTE2_SPIM2_SPIS2_TWIM2_TWIS2);
```

It gives the following error:

```txt
error[E0433]: failed to resolve: use of undeclared crate or module `interrupt`
  --> secure/src/main.rs:21:15
   |
21 |     let irq = embassy_nrf::interrupt::take!(UARTE2_SPIM2_SPIS2_TWIM2_TWIS2);
   |               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ use of undeclared crate or module `interrupt`
   |
   = note: this error originates in the macro `embassy_nrf::interrupt::take` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: use of undeclared crate or module `interrupt`
  --> secure/src/main.rs:21:15
   |
21 |     let irq = embassy_nrf::interrupt::take!(UARTE2_SPIM2_SPIS2_TWIM2_TWIS2);
   |               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ use of undeclared crate or module `interrupt`
   |
   = note: this error originates in the macro `embassy_nrf::interrupt::take` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing this struct
   |
7  | use core::sync::atomic::AtomicBool;
   |
```

It does work when I have the following code:

```rust
use embassy_nrf::interrupt;
let irq = interrupt::take!(UARTE2_SPIM2_SPIS2_TWIM2_TWIS2);
```

Seems like the `take!` macro makes an assumption about the existing imports.

Dirbaio, 2022-11-08:
The original `cortex-m-rt`'s `#[interrupt]` also has the same problem! It's so that it can "check" the interrupt actually exists, it searches for it in the `interrupt` namespace, so it has to be imported... 

I agree it's very confusing, but I don't know a way to fix it without worsening the API :(

diondokter, 2022-11-08:
Right, I had it before with `cortex-m-rt` that I spent an hour just trying to figure out what's going wrong. The error is pretty confusing yeah.

What I don't get is why it couldn't just use `embassy_nrf::interrupt` inside the macro

Dirbaio, 2022-11-08:
> What I don't get is why it couldn't just use embassy_nrf::interrupt inside the macro.

Oh, hmmm. The reason it doesn't currently is because the macro is shared for all HALs, so it doesn't "know" it's being used from `embassy_nrf`. 

It could be done by creating 1 macro in `embassy-macros` for each HAL (that can all call some shared code) and then reexporting the right one from each HAL. Not pretty, but doable.

Dirbaio, 2023-03-14:
This is fixed with the new interrupt system: #1224. Thanks to being a proc macro, it can use `$crate`, so it always works no matter how you import it.

closed 2023-03-14 by Dirbaio