embassy-rs/embassy #1050: Embassy release TODO

lulf, 2022-11-10:
This is an attempt at capturing remaining work needed for crate releases. Feel free to edit/propose changes.

* embassy-executor: ready!

* embassy-cortex-m
** [ ] pending cortex-m-rt interrupt model work (@Dirbaio maybe expand on this)
** [ ] document public APIs

* embassy-net
** [ ] document public APIs

* embassy-usb
** [ ] document public APIs
** [ ] dependencies: embassy-usb-driver

* embassy-usb-driver
** [ ] document public APIs

* embassy-hal-common
** [ ] document public APIs
* embassy-embedded-hal
** [ ] document public APIs
* embassy-nrf
** [ ] dependencies: embassy-cortex-m, embassy-hal-common, embassy-embedded-hal, embassy-usb-driver
** [ ] document public APIs
** [ ] #980 - remove instance generics
* embassy-stm32
** [ ] dependencies: embassy-cortex-m, embassy-hal-common, embassy-embedded-hal, embassy-usb-driver, embassy-net
** [ ] #980 - remove instance generics
** [ ] document public APIs

* embassy-lora
** [ ] dependencies: embassy-stm32, embassy-hal-common

* embassy-boot: ready!
* embassy-boot-nrf
** dependencies: embassy-nrf, embassy-boot
* embassy-boot-stm32
** dependencies: embassy-stm32, embassy-boot

elpiel, 2023-05-01:
It would be nice to add `embassy-rp` to the release tasks too.

JuliDi, 2023-10-12:
What's the preferred way of documentation w.r.t. this planned release?
Single function documentation as docstrings, sure.
But for bigger "how to" guides or MWEs, should they go on the website, docstrings at the module level or as per-chip examples (like we currently have but with more documentation) or a"book"?

I'd be open to documenting more stuff as I progress in my projects using embassy. Having a stable release soon-ish would be great.

elpiel, 2024-01-26:
@lulf many of the crates were recently released on crates.io.
Could we update this issue to reflect this and see if there are any remaining tasks (crates to be released)?

lulf, 2024-01-26:
All is released from what I can tell, closing.

closed 2024-01-26 by lulf