embassy-rs/embassy #1061: Executor is not fair if spi dma transfer are used

MauroMombelli, 2022-11-20:
Minimal example to reproduce the issue follow.

EXPECTED:
START A and START B are both printed (order can be inverted)
A and B are printed, in different order/frequency

OBSERVED:
START A and A are NEVER printed.
START B and B are always printed.

FUN FACT:
Adding a third task, is also never run. "spi.transfer().await;" seems to monopolize the executor
Adding a little delay with "Timer::after(Duration::from_millis(100)).await;"  to the spi's task will let START A and A get printed

CODE:

`#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]

use defmt::*;
use embassy_executor::Spawner;
use embassy_futures::yield_now;
use embassy_stm32::{spi::{Config, Spi}, peripherals::{SPI1, DMA1_CH3, DMA1_CH2}};

use {defmt_rtt as _, panic_probe as _};

// Declare async tasks
#[embassy_executor::task]
async fn a() {
    info!("START A");
    
    loop {
        info!("A");
        yield_now().await;
    }
}

#[embassy_executor::task]
async fn spi_poll(mut spi: Spi<'static, SPI1, DMA1_CH3, DMA1_CH2>) {
    info!("START B");
    
    loop {
        info!("B");
        let write: [u8; 7] = [0; 7];
        let mut read: [u8; 7] = [0; 7];
        let _ = spi.transfer(&mut read, &write).await;
    }
}

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    
    let p = embassy_stm32::init(Default::default());

    info!("Hello World!");

    spawner.spawn(a()).unwrap();
    
    let spi = Spi::new(
        p.SPI1,
        p.PA5,
        p.PA7,
        p.PA6,
        p.DMA1_CH3,
        p.DMA1_CH2,
        embassy_stm32::time::Hertz(1_000_000),
        Config::default(),
    );

    spawner.spawn(spi_poll(spi)).unwrap();
}
`

yodaldevoid, 2022-11-26:
I have also run into this issue, but when using `embassy_rp` instead. Adding a `yield_now` to my SPI task also addressed the problem.

nikvoid, 2023-03-26:
This is happening because without `yield_now().await` you just do not have any await-points in `a` task, so it blocks entire executor in this endless loop
```
    loop {
        info!("A");
        yield_now().await; // await point, here executor can switch tasks
    }
```

lestofante, 2023-03-26:
@nikvoid read the issue again, you are completely missing what is going on here

nikvoid, 2023-03-26:
Oh, get it wrong, sorry

JonathanLindsey, 2025-02-19:
Even with #1474 merged this is still an issue. Or at least there is a similar issue with spi.write() .

There are additional conditions:
1. Must use `spi.write()` instead of `spi.transfer()`.
2. The buffer to write must have a length of 9 or less.
3. OR using `spi.transfer()` the length of the buffer to write must be 0. ;-)

It doesn't seem to me that using yield_now() is a perfect solution since spi.write() is still blocking, although, only for a short amount of time.

In the same vein as the example above, here is an example. Comment/uncomment the appropriate lines and change the buffer sizes to test the various additional conditions I outlined. I recommend to comment out the `info!("B");` line to make it more obvious what is going on.

```rust
#![no_std]
#![no_main]

use defmt::*;
use embassy_executor::Spawner;
use embassy_futures::yield_now;
use embassy_rp::{
    peripherals::SPI0,
    spi::{self, Config, Spi},
};
// use embassy_stm32::{
//     peripherals::{DMA1_CH2, DMA1_CH3, SPI1},
//     spi::{Config, Spi},
// };

use {defmt_rtt as _, panic_probe as _};

// Declare async tasks
#[embassy_executor::task]
async fn a() {
    info!("START A");

    loop {
        info!("A");
        yield_now().await;
    }
}

#[embassy_executor::task]
async fn spi_poll(mut spi: Spi<'static, SPI0, spi::Async>) {
    info!("START B");

    loop {
        info!("B");
        let write = [0; 9];
        let mut read = [0; 9];
        // spi.transfer(&mut read, &write).await.unwrap();
        spi.write(&write).await.unwrap();
    }
}

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let p = embassy_rp::init(Default::default());

    info!("Hello World!");

    spawner.spawn(a()).unwrap();

    let spi = Spi::new(
        p.SPI0,
        p.PIN_2,
        p.PIN_3,
        p.PIN_4,
        p.DMA_CH0,
        p.DMA_CH1,
        Config::default(),
    );

    spawner.spawn(spi_poll(spi)).unwrap();
}
```