embassy-rs/embassy #1062: stm32f303 i2c.write_read can hang

MauroMombelli, 2022-11-20:
In my case write_read is using DMA, but even the DMA version is calling master_write WITHOUT timeout.
This causes the whole system to hang (seems like a problem in i2c in those chip, they fail on generation of start).

I tried wrapping into a with_timeout, but it seems to not have effect

```
let fut = i2c.write_read(acce_addr, &buf_read_request_i2c, &mut buf_read_i2c);
if let Ok(_) = with_timeout(Duration::from_millis(5), fut).await {
```

call stack:
```
embassy_stm32::i2c::<impl embassy_stm32::i2c::sealed::Instance for embassy_stm32::_generated::peripherals::I2C1>::regs@0x0800326e (Unknown Source:0)
embassy_stm32::i2c::_version::I2c<T,TXDMA,RXDMA>::master_write@0x08000290 (Unknown Source:0)
<core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll@0x08001312 (Unknown Source:0)
<core::future::from_generator::GenFuture<T> as core::future::future::Future>::poll@0x08002420 (Unknown Source:0)
_ZN16embassy_executor3raw20TaskStorage$LT$F$GT$4poll17h61388c2a994bf1e3E.llvm.3235118994076707623@0x08002890 (Unknown Source:0)
embassy_executor::raw::Executor::poll@0x08004e02 (Unknown Source:0)
embassy_executor::arch::Executor::run@0x080010ea (Unknown Source:0)
robottonzo::__cortex_m_rt_main@0x08001068 (Unknown Source:0)
main@0x08001052 (Unknown Source:0)
```

also it seems to be happening more often in release rather than debug

nathanaelg, 2026-09-21:
Did you have any progress with this? I seem to be having a similar issue with STM32U575

MauroMombelli, 2026-09-28:
I think is related to https://github.com/embassy-rs/embassy/issues/1061
ill look into it soonish as im working on some code recently, just not there yet