embassy-rs/embassy #1070: Implement a shareable struct between taks

miathedev, 2022-11-22:
Hi fellow ambassador Rustoceans crab !

I hardly tried to implement a shareable struct between embassy tasks.
I want to share a struct like this:

```
impl SomeObj {
    pub fn new() -> Self {
        Self { counter: 0 }
    }

    pub fn say_hello(&mut self) {
        defmt::info!("Hello Threads: {}", self.counter);
    }
}
```

Is there a easy way, to share this structs instance, between tasks using no_std?
My goal was to actually create an embassy example code for this standard issue.
But i honestly failed.

I came up this idea, because i wanted to implement a little reusable driver for printing to a UART Periph within multiple tasks.

With kind regards,
Mia

MauroMombelli, 2022-11-23:
all you need is in embassy_sync, Signal, Mutex and Channel.
For help question is faster and easier if you ask in chat (https://matrix.to/#/#embassy-rs:matrix.org), i guess here is only issue for bug reporting

lulf, 2022-11-23:
Asking questions here is totally fine, but probably the chat will be more responsive :) For examples, have a look at 

https://github.com/embassy-rs/embassy/blob/master/examples/nrf/src/bin/pubsub.rs
https://github.com/embassy-rs/embassy/blob/master/examples/nrf/src/bin/mutex.rs
https://github.com/embassy-rs/embassy/blob/master/examples/nrf/src/bin/channel.rs

closed 2023-03-14 by Dirbaio