embassy-rs/embassy #1080: Spawner: Two-staged spawn

chrysn, 2022-11-28:
When preparatory work for a pooled task to spawn is costly, it'd be great to get commitment from the spawner to have a slot prepared that I can then fill the spawn token into.

In usage, this might look like this:

```rust
if let Ok(task) = spawner.prepare_spawn() { /* type inferred */
    let con = peripheral::advertise_connectable(...).await;
    task.spawn(blueworker(server, con)); /* infallible now */
}
```

As the example illustrates, this would come handy with softdevice connections (when their number is synchronized with the task pool).

I know little of embassy internals, but I guess that prepare_spawn might increment the task counter without actually putting that task on the run list; spawn would do that, and dropping the prepared object without spawning it decrements the counter.

Currently, as a workaround, I keep track of the number of tasks in the pool manually, but this is a) tedius, and b) inherently racy, because decrementing the number of running tasks happens in the task itself, and in a multithreaded environment (which AIU this is not, so it's more a matter of style).

Dirbaio, 2022-11-28:
This is very interesting. I've needed a feature like this myself at times, having it would be nice.

I'm not sure what the API should look like though. `spawner.prepare_spawn()` looks like reserving capacity in the `spawner`, while it's reserving it in the task's pool. Relying on type inference to make that reserve from the right pool seems a bit too magic IMO.

I think the line of code that does the reserve should have the task name. But how...? Something like this:

```rust
if let Ok(task) = blueworker::prepare_spawn() {
    let con = peripheral::advertise_connectable(...).await;
    spawner.spawn(task(server, con)); /* infallible now */
}
```

but having the macro turn `async fn blueworker()` into something that looks like a mod (`blueworker::foo()`) seems like too much magic too....

chrysn, 2022-11-28:
I was under the (possibly unwarranted) impression that both the pool and the signature is part of the type, so `.prepare()` would know what to act on just like `.spawn()` knows which pool to spawn from by the type (and not instance, because the instance merely encodes the parameters) of the argument, but I'd be just as happy with doing this differently. 

Might it be possible to leverage an extension of the task invocation? Maybe

```
#[embassy_executor::task(pool_size=2, pool_name=BLUEWORKERS)]
async fn blueworker(...) { }

... {

    if let Some(task) = spawner.prepare_spawn(BLUEWORKERS) {
        ...
        task.spawn(blueworker(server, con));
    }
}
```

An explicit name might also be useful when implementing something `top` like that reports that

```
println!("Used Bluetooth tasks: {} out of {}", BLUEWORKER.Stasks_active(), BLUEWORKERS.tasks_max);
```

(or `spawner.tasks_active(BLUEWORKERS)` depending on where the state sits)

Dirbaio, 2022-11-28:
> so .prepare() would know what to act on just like .spawn() knows which pool to spawn from by the type (and not instance, because the instance merely encodes the parameters) of the argument

The way it works internally right now is: the task is allocated from the pool the moment you do `blueworker(server, conn)`. That returns a `SpawnToken` which is just a pointer. `spawn` has no idea which pool it's spawning from. (If allocation fails, it returns a "poisoned" `SpawnToken` which causes `spawn` to return the error.)

The reason for the weird "poison the spawntoken, return the error later from `spawn`" was to mirror the API of "traditional" executors... which is a bit pointless perhaps because `spawn` is infallible in those.

Perhaps that was a mistake in retrospect, and we should change the macro to "declare a taskpool" that has a richer api, yes...

chrysn, 2026-10-20:
I think that with the changes in embassy-executor, where the creation of the SpawnToken is now fallible.

It's not all solved, because I think I'd still need to get more data into the nascent task, but it looks easier now.