embassy-rs/embassy #1094: stm32 HAL DMA SPI CLK pin floats during start of transfer

JoshMcguigan, 2022-12-05:
## Problem description

The embassy stm32 HAL for DMA SPI lets the clock line float briefly during the setup of a transaction, rather than continuing to drive it (it should be driven low, in the configuration shown here). This results in the sensor reading errant data (see below). The MOSI line also floats during this time, but that is less of a problem.

## Reproduction

```
#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let p = embassy_stm32::init(Default::default());

    let spi_bus_1 = {
        let sck = p.PA5;
        let miso = p.PA6;
        let mosi = p.PA7;
        let spi = Spi::new(
            p.SPI1,
            sck,
            mosi,
            miso,
            p.DMA2_CH3,
            p.DMA2_CH0,
            mhz(1),
            spi::Config::default(),
        );
        SPI_BUS.init(Mutex::<ThreadModeRawMutex, _>::new(spi))
    };
    let csn = Output::new(p.PB6, Level::High, Speed::High);
    let mut spi_dev_1 = SpiDevice::new(spi_bus_1, csn);

    let mut buf = [0u8, 0];
    embedded_hal_async::spi::SpiDevice::transfer_in_place(&mut spi_dev_1, &mut buf).await.unwrap();
    println!("{:?}", buf);
}
```

## Relevant code

There may not be an easy fix for this, since control of the chip select line is handled outside this HAL code - and the stm32 reference manual says you are required to disable SPI before setting up a DMA transfer. The specific line that lets the pin start floating is below:

https://github.com/embassy-rs/embassy/blob/5e94b8060b7f1af8118ca1f943181ce70be92057/embassy-stm32/src/spi/mod.rs#L512

## Potential solutions

A few solutions to this were discussed in the matrix chat.

- Keep the DMA transaction open across calls to read/write
  - This may be problematic if the user changes bus settings between calls, and also the reference manual says the chip should not be placed into low power mode while SPI is enabled
- Set the CLK (and maybe MOSI) pins as pull-down on setup (or pull-up, depending on config - and this would need to be updated if the user modified the config)
  - This is less good than driving the pin to the correct value, but may be better than nothing
- Document this and require users fix it themselves (add a pull-up/down resistor - or configure the pins as pull-up/pull-down before passing them into SPI setup)
- Dig around in the reference manual and determine if there is a better way to start/stop a DMA transaction while keeping active control of the clock the whole time
  

## Extra data

Expected (taken from stm32f4xx-hal)

[image]

Embassy stm32 HAL w/ DMA (note the long CLK "pulse" at the beginning)

[image]

noppej, 2023-01-12:
I think I get the same behaviour. I don't have a logic analyzer to confirm, but my driver works perfectly, and then all of a sudden starts sending garbage data. Removing power from the STM32H745 controller and SPI slave will reset it and it will work fine for a while again, until it starts again. It gets worse when I use DMA in the config of my SPI3 ... then my SPI is permanently garbage. Is that because of #1095?

jr-oss, 2023-02-05:
I encountered the same behavior with an stm32l4.
In RM0394 it states:

> 40.4.6
Communication formats
> ...
> The idle state of SCK must correspond to the polarity selected in the SPIx_CR1 register (by
pulling up SCK if CPOL=1 or pulling down SCK if CPOL=0).

This can be achieved as already described in the potential solutions by setting pull-up/-down mode when initializing SPI:
https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/spi/mod.rs#L97

For my tests  with SPI MODE3 I got it working by replacing
`sck.set_as_af(sck.af_num(), AFType::OutputPushPull);`
by
`sck.set_as_af_pull(sck.af_num(), AFType::OutputPushPull, Pull::Up);`

closed 2023-02-07 by bors[bot]

JoshMcguigan, 2023-02-09:
Just confirming this is fixed for me now, with #1199. Thanks @jr-oss!

commit c4a2c620 (2023-02-07): Merge #1199

1199: STM32 SPI: Set clk-pin pull-up/-down to match spi clock polarity r=Dirbaio a=jr-oss

Fixes #1094 

There are some proposed solutions in #1094 

> Keep the DMA transaction open across calls to read/write
        This may be problematic if the user changes bus settings between calls, and also the reference manual says the chip should not be placed into low power mode while SPI is enabled

As already described, this is problematic and against reference manual recommendation

>    Set the CLK (and maybe MOSI) pins as pull-down on setup (or pull-up, depending on config - and this [... cut]
PR #1199 (2023-02-07): STM32 SPI: Set clk-pin pull-up/-down to match spi clock polarity
Fixes #1094 

There are some proposed solutions in #1094 

> Keep the DMA transaction open across calls to read/write
        This may be problematic if the user changes bus settings between calls, and also the reference manual says the chip should not be placed into low power mode while SPI is enabled

As already described, this is problematic and against reference manual recommendation

>    Set the CLK (and maybe MOSI) pins as pull-down on setup (or pull-up, depending on config - and this would need to be updated if the user m [... cut]