embassy-rs/embassy #1095: stm32 HAL DMA SPI read broken

JoshMcguigan, 2022-12-05:
Performing a transfer in place correctly reads data from the sensor. Doing the equivalent with a write and then read sends/receives the same bytes over the wire, but the `read` call returns the wrong data.

```
    // In my tests this call was actually done twice, as shown in the logic analyzer trace below
    let mut buf = [0u8, 0];
    embedded_hal_async::spi::SpiDevice::transfer_in_place(&mut spi_dev_1, &mut buf).await.unwrap();
    println!("{:?}", buf);

    let mut buf = [0u8];
    embedded_hal_async::spi::transaction!(&mut spi_dev_1, move |bus| async move {
        bus.write(&buf).await?;
        bus.read(&mut buf).await
    }).await.unwrap();
    println!("{:?}", buf);
```

```
// Output (note the third read, done with a write then read, returns 0 rather than 8)

0.000213 [14, 8]
└─ nucleo_f446re_embassy::____embassy_main_task::{async_fn#0}::{closure#0} @ src/main.rs:62
0.000549 [14, 8]
└─ nucleo_f446re_embassy::____embassy_main_task::{async_fn#0}::{closure#0} @ src/main.rs:66
0.000915 [0]
└─ nucleo_f446re_embassy::____embassy_main_task::{async_fn#0}::{closure#0} @ src/main.rs:73
```

Note each transaction sends/receives the same bits:

[image]

rmja, 2022-12-22:
I have the same problem. I will try and investigate and get back.

rmja, 2022-12-23:
@JoshMcguigan Could you verify that my PR resolves the issue?

JoshMcguigan, 2022-12-25:
This issue (or something similar, write then read returns different data compared to transfer in place) still shows up for me on the latest master branch. I don't have time at the moment to hook up the logic analyzer unfortunately.

Redrield, 2023-02-01:
Ran into this on 7e251a25, trying to interface with the gyro on the STM32F429I-DISC1. Grabbed a video of what the bus looks like from a logic analyzer, with `spi.write` and `spi.read` back to back in Embassy, vs an STM32Cube based application polling the same register. When I disable rx DMA, embassy's behaviour matches that of Cube

https://user-images.githubusercontent.com/8991639/216128314-610c1785-8ac2-4443-b575-1d487c309d9e.mp4

https://user-images.githubusercontent.com/8991639/216128365-be6a8730-8506-48e5-9188-c40c81dbf45a.mp4

Dirbaio, 2024-06-02:
i'm going to close this since there's no response from OP and a lot of time has passed. I've never been able to reproduce this and there's been many SPI fixes, so it's likely it's either fixed already or the cause was unrelated to embassy. Feel free to reopen if it still happens and you have oscilloscope traces.

commit 662bb579 (2022-12-23): fix(stm32): Ensure that gpio speed is VeryHigh for all spi versions

This fixes #1095
commit 40ef66cd (2022-12-23): Merge #1124

1124: Fix two SPI bugs for stm32 r=Dirbaio a=rmja

This PR fixes two bugs:
* It fixes #1095 by ensuring that pin speed is VeryHigh for all spi versions. I am on stm32f429 which seems to be spi_v1, and it also needs the VeryHigh pin speed. Otherwise bit errors on the "last bit in every byte" can happen.
* It also fixes a lifetime bug for the tx buffer when sending "write_repeated". The issue can be seen when doing spi.write where the clock byte changes during a transmission because the buffer handled to the dma must live throughout the entire transfer.
PR #1124 (2022-12-23): Fix two SPI bugs for stm32
This PR fixes two bugs:
* It fixes #1095 by ensuring that pin speed is VeryHigh for all spi versions. I am on stm32f429 which seems to be spi_v1, and it also needs the VeryHigh pin speed. Otherwise bit errors on the "last bit in every byte" can happen.
* It also fixes a lifetime bug for the tx buffer when sending "write_repeated". The issue can be seen when doing spi.write where the clock byte changes during a transmission because the buffer handled to the dma must live throughout the entire transfer.
closed 2024-06-02 by Dirbaio