embassy-rs/embassy #1106: embassy-net should also account for alloc feature

stevefan1999-personal, 2022-12-10:
Although I know it is very much not recommended because the memory budget can vary over builds on IoT devices, I wanted to experiment with https://github.com/yvt/rlsf to have a fun experiment on how using much the memory fragmentation, performance and energy impact can be done by using this kind of heap memory. It also serves for future use of SPI-based external RAM, since we don't really need fast RAM for storing socket buffers.

Right now if you enabled `alloc` on smoltcp altogether with `embassy-net` it will cause an error like this:
```
error[E0004]: non-exhaustive patterns: `&mut ManagedSlice::Owned(_)` not covered                                                                                                                                                                                                                                       
  --> C:\Users\steve\.cargo\git\checkouts\smoltcp-d6a83fd2b0891566\b7a7c4b\src\iface\fragmentation.rs:79:15
   |
79 |         match &mut self.buffer {
   |               ^^^^^^^^^^^^^^^^ pattern `&mut ManagedSlice::Owned(_)` not covered
   |
note: `ManagedSlice<'_, u8>` defined here
  --> C:\Users\steve\.cargo\registry\src\github.com-1ecc6299db9ec823\managed-0.8.0\src\slice.rs:34:5
   |
29 | pub enum ManagedSlice<'a, T: 'a> {
   | --------------------------------
...
34 |     Owned(Vec<T>)
   |     ^^^^^ not covered
   = note: the matched value is of type `&mut ManagedSlice<'_, u8>`
help: ensure that all possible cases are being handled by adding a match arm with a wildcard pattern or an explicit pattern as shown
   |
86 ~             }
87 +             &mut ManagedSlice::Owned(_) => todo!()
   |

error[E0004]: non-exhaustive patterns: `&mut ManagedSlice::Owned(_)` not covered
   --> C:\Users\steve\.cargo\git\checkouts\smoltcp-d6a83fd2b0891566\b7a7c4b\src\iface\fragmentation.rs:161:23
    |
161 |                 match &mut self.buffer {
    |                       ^^^^^^^^^^^^^^^^ pattern `&mut ManagedSlice::Owned(_)` not covered
    |
note: `ManagedSlice<'_, u8>` defined here
   --> C:\Users\steve\.cargo\registry\src\github.com-1ecc6299db9ec823\managed-0.8.0\src\slice.rs:34:5
    |
29  | pub enum ManagedSlice<'a, T: 'a> {
    | --------------------------------
...
34  |     Owned(Vec<T>)
    |     ^^^^^ not covered
    = note: the matched value is of type `&mut ManagedSlice<'_, u8>`
help: ensure that all possible cases are being handled by adding a match arm with a wildcard pattern or an explicit pattern as shown
    |
166 ~                     }
167 +                     &mut ManagedSlice::Owned(_) => todo!()
    |

```

Dirbaio, 2022-12-11:
This happens when you enable `alloc` on `managed` and not on `smoltcp`.

Note that `embassy-net` uses a particular [smoltcp git version](https://github.com/embassy-rs/embassy/blob/5d4f09156af094732edc5c01332af8b13db10e0f/embassy-net/Cargo.toml#L60), so make sure to use that exact one to enable the `alloc` feature. Otherwise `cargo` builds the 2 versions of smoltcp, enabling `alloc` only in yours.

Dirbaio, 2023-03-14:
Closing since this is a smoltcp issue, not embassy's.

closed 2023-03-14 by Dirbaio