embassy-rs/embassy #1107: Correct abstraction layer for USB access

stillinbeta, 2022-12-10:
I'm working on adding USB support to `stm32g0[bc]1`, and I've encountered a bit of a fork.

Because these chips support USB Host mode, a lot of the functions have different names. For example, [EPR](https://docs.embassy.dev/stm32-metapac/git/stm32g474vc/usb/struct.Usb.html#method.epr) is `CHEPR`, [FSUSP](https://docs.embassy.dev/stm32-metapac/git/stm32g474vc/usb/regs/struct.Cntr.html#method.fsusp) is `SUSPEN`, etc. 

My question is: should I try to modify [embassy_stm32::usb::usb](https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/usb/usb.rs) to accommodate these renames, in the form of a macro a la  [the stm32xx_hal](https://github.com/stm32-rs/stm32g0xx-hal/blob/main/src/timer/pwm.rs#L294-L299) crates? Or would it be better to just rename the fields in my [usb_v4.yaml](https://github.com/embassy-rs/stm32-data/pull/161/files#diff-6053a776ef729d59a13bdd2ec70d7266d5dab4f9a1ae54c901b0e8b5e28ad1c6)?

stillinbeta, 2022-12-11:
here's the rough mapping 
register | existing name | g0 name
-- | -- | --
  | EPRn | CHEPRn
CHEPRn | CTR_RX | VTRX
  | EP_MEM | not real (?)
USB_CNTR | FRES | USBRST
  | USB_BTABLE | (always zero)
USB_CNTR | FSUSP | SUSPEN
USB_CNTR | LP_MODE | part of SUSPEN
USB_ISTR | RESET | RST_DCON

Dirbaio, 2022-12-11:
> Or would it be better to just rename the fields in my [usb_v4.yaml](https://github.com/embassy-rs/stm32-data/pull/161/files#diff-6053a776ef729d59a13bdd2ec70d7266d5dab4f9a1ae54c901b0e8b5e28ad1c6)?

This.

The goal of stm32-data / stm32-metapac is to allow easily writing a HAL that supports all stm32 families. If ST itself is inconsistent with the naming, we "fix" it for them :) 

Otherwise the HAL would get full of `cfg`s to accomodate all the inconsistencies, which quickly becomes unmaintainable.

Dirbaio, 2023-03-14:
Closing since G0 usb is already done.

closed 2023-03-14 by Dirbaio

stillinbeta, 2023-03-14:
oh excellent, glad to hear it's been done! Is there a PR I could look at?

Dirbaio, 2023-03-14:
only thing I can find is yours https://github.com/embassy-rs/stm32-data/pull/161

but it seems everything's already building fine in the embassy-stm32 side, usb is there on G0 chips. So I guess it should be working already?

daehyeok, 2023-07-17:
Hi,
 Anyone verified that is it working? 

 I'm trying to use STM32G0B1KET6 chip on my custom board.
for the quick test, I tried to run modified USB-Serial Example like below gist.

https://gist.github.com/daehyeok/8aad353e9e2d7fa1d73b83b07448b6aa

 build and flash success without any issue. but any usb device is not appeared on my M2 mac. (I checked "DFU in FS mode" is on System infomration before flash firmware.  so i think hardware is okay).

stillinbeta, 2023-07-17:
As far as I know, there's no support for STM32G0B1s in Embassy, and it'll require a nearly from-scratch driver. It's on my personal list, but obviously it's a huge undertaking and I don't have any estimates.

daehyeok, 2023-07-17:
@Dirbaio Thanks for reply.
According to application note [AN5096](https://www.st.com/resource/en/application_note/an5096-getting-started-with-stm32g0-series-hardware-development-stmicroelectronics.pdf), USB use HSI48, but i couldn't find any related config on [rcc/g0.rs](https://github.com/embassy-rs/embassy/blob/main/embassy-stm32/src/rcc/g0.rs). is it missing parts?

daehyeok, 2023-07-18:
update 

I turn on HSI48 directly through pac
```rust
pac::RCC.cr().modify(|w| w.set_hsi48on(true));
pac::RCC.ccipr2().modify(|w| w.set_usbsel(pac::rcc::vals::Usbsel::HSI48));
```

Then usb_serial and HID working well!

stillinbeta, 2023-08-07:
Do you happen to have a gist or an example repo? I haven't been able to get
mine working

daehyeok, 2023-08-07:
Here is my keyboard firmware repo with STM32G0B1KET6

https://github.com/daehyeok/corne-eec-rs