embassy-rs/embassy #1108: Docs build features

elpiel, 2022-12-11:
I found a few things in the docs and wanted to report them (I could look into fixing them):

- [ ] `embassy-net` does not build with the `udp` feature (the only feature not included in the docs)

- [ ] https://docs.embassy.dev/embassy-lora/git/stm32wl/stm32wl/struct.SubGhzRadioConfig.html has fields that point to bad links like: `https://docs.embassy.dev/__DOCSERVER_DEPLINK/embassy-stm32/embassy_stm32/subghz/reg_mode/enum.RegMode.html` which is not a valid link to the docs of `embassy-stm32` crate. This could be because `embassy-stm32` docs are not built with the `subghz` feature and `Cargo.toml` says it's a todo:

https://github.com/embassy-rs/embassy/blob/5d4f09156af094732edc5c01332af8b13db10e0f/embassy-stm32/Cargo.toml#L7-L14

ceekdee, 2023-04-30:
SubGhzRadioConfig is no longer referenced after the removal of the original stm32wl LoRa driver in embassy-lora as part of PR https://github.com/embassy-rs/embassy/pull/1407.

Dirbaio, 2023-05-01:
True! I'm leaving this issue open because it actually affects all links to deps that are not on docs.rs.

Dirbaio, 2023-12-13:
intra-crate links are now fixed.

i'm still not sure what's up with the embedded-hal versions, it might be a rustdoc bug?

closed 2023-12-13 by Dirbaio