embassy-rs/embassy #1110: example: std net fail on Ubuntu 22.10

overheat, 2022-12-12:
I follow `embassy-net/README.md` instructions and got fail on Ubuntu 22.10
```
$ ip addr show
...
13: tap0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether c6:95:7b:04:4a:5c brd ff:ff:ff:ff:ff:ff
    inet 192.168.69.100/24 scope global tap0
       valid_lft forever preferred_lft forever
    inet6 fdaa::100/64 scope global tentative 
       valid_lft forever preferred_lft forever
    inet6 fe80::100/64 scope link tentative 
       valid_lft forever preferred_lft forever
```

```
$ cargo run --bin net
    Finished dev [unoptimized + debuginfo] target(s) in 0.12s
     Running `target/debug/net`
[2022-12-12T08:51:36.626714025Z INFO  net] connecting to (Address([192, 168, 69, 100]), 8000)...
[2022-12-12T08:51:36.626872456Z INFO  embassy_net::stack] link_up = true
[2022-12-12T08:51:36.626894708Z DEBUG embassy_net::stack] Lost IP configuration
[2022-12-12T08:51:46.627333276Z WARN  net] connect error: ConnectionReset
^C
```

Dirbaio, 2022-12-12:
It connects to TCP port 8000, you need to have something listening there. For example `nc -l 8000`

Dirbaio, 2022-12-13:
Fixed in #1111
Thanks!

closed 2022-12-13 by Dirbaio