embassy-rs/embassy #1112: How to test example nrf/usb_ethernet?

overheat, 2022-12-15:
My PC, Ubuntu, recognized the USB Ethernet card, but no IPv4 address after I change `examples/nrf/src/bin/usb_ethernet.rs` like this:
```
    // let config = embassy_net::ConfigStrategy::Dhcp;
    let config = embassy_net::ConfigStrategy::Static(embassy_net::Config {
            address: Ipv4Cidr::new(Ipv4Address::new(192, 168, 69, 2), 24),
            dns_servers: Vec::new(),
            gateway: Some(Ipv4Address::new(192, 168, 69, 1)),
    });
```

```
$ ip addr show
...
34: enx888888888888: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 88:88:88:88:88:88 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::9e5f:fc98:3cc5:3728/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
```
```
$ lsusb
...
Bus 001 Device 009: ID 1366:1015 SEGGER J-Link
Bus 001 Device 019: ID c0de:cafe Embassy USB-Ethernet example
...
```

```
$ ls -l /dev/bus/usb/001/019
crw-rw-r-- 1 root root 189, 18 12月 15 15:28 /dev/bus/usb/001/019
```

Dirbaio, 2022-12-15:
The device acts as a DHCP client, so you'll need something to act as a DHCP server on that interface. The easiest is to set the connection to "Shared with other computers" in NetworkManager. This'll cause the device to obtain an IP, at which point you can ping or talk to it.

overheat, 2022-12-15:
> // let config = embassy_net::ConfigStrategy::Dhcp;
>     let config = embassy_net::ConfigStrategy::Static(embassy_net::Config {
>             address: Ipv4Cidr::new(Ipv4Address::new(192, 168, 69, 2), [...]

I think I already change the device from DHCP client to static IP by modify the source code like this:
```
    // let config = embassy_net::ConfigStrategy::Dhcp;
    let config = embassy_net::ConfigStrategy::Static(embassy_net::Config {
            address: Ipv4Cidr::new(Ipv4Address::new(192, 168, 69, 2), 24),
            dns_servers: Vec::new(),
            gateway: Some(Ipv4Address::new(192, 168, 69, 1)),
    });
```

Dirbaio, 2022-12-15:
If you want to use static IP, you have to manually configure the interface on the Linux side with a different static IP in the same subnet.

overheat, 2022-12-17:
> The device acts as a DHCP client, so you'll need something to act as a DHCP server on that interface. The easiest is to set the connection to "Shared with other computers" in NetworkManager. This'll cause the device to obtain an IP, at which point you can ping or talk to it.

I still try to finger out how to `set the connection to "Shared with other computers" in NetworkManager` on my Ubuntu 22.10, when the device acts as a DHCP client. @Dirbaio Could you give me some clues?

Dirbaio, 2022-12-18:
In ipv4 settings you have to select "Shared to other computers": https://fedoramagazine.org/internet-connection-sharing-networkmanager/

overheat, 2022-12-21:
> The device acts as a DHCP client, so you'll need something to act as a DHCP server on that interface. The easiest is to set the connection to "Shared with other computers" in NetworkManager. This'll cause the device to obtain an IP, at which point you can ping or talk to it.

Yes, change this `Eembassy Ethernet` card to `Share to other computers` in Ubuntu Settings->Network. Then re-enable it.

The IP address will change from `169.254.x.x` to `10.x.x.x`, which is allocated for private internets. 
```
$ ping 10.42.0.1
PING 10.42.0.1 (10.42.0.1) 56(84) bytes of data.
64 bytes from 10.42.0.1: icmp_seq=1 ttl=64 time=0.034 ms
64 bytes from 10.42.0.1: icmp_seq=2 ttl=64 time=0.030 ms
...
```
But another error on my Pico W running `examples/rp/src/bin/usb_ethernet.rs`
```
$ socat stdio tcp4-connect:10.42.0.1:1234  <<<"abcdefg"
2022/12/21 15:25:57 socat[7385] E connect(5, AF=2 10.42.0.1:1234, 16): Connection refused
```

overheat, 2022-12-21:
Resolved!

Looks `ip addr show` tell me the `server_ip: 10.42.0.1`, device ip is `your_ip: 10.42.0.232`

you can find it in the logger output.

@Dirbaio Thanks again.

closed 2022-12-21 by overheat

overheat, 2022-12-21:
Update:
On my Ubuntu 22.10, look need `sudo ufw disable` to let device get IP address through DHCP when testing USB Ethernet example.