embassy-rs/embassy #1115: duplicate of embassy-time causes timers to have wrong wall time

TheButlah, 2022-12-16:
Problem:
If your `embassy-time` is duplicated in your dependency graph, such as if cargo was using both the git version and the crates.io version at the same time, delays have the wrong wall time. For example, a 100ms delay beecomes a 3-4 second delay.

Minimally reproducible example: https://github.com/SlimeVR/SlimeVR-Rust/blob/cfd00854a08e65fcb4d2946628ba4438531b829f/firmware/src/main.rs

Link to matrix discussion: https://matrix.to/#/!YoLPkieCYHGzdjUhOK:matrix.org/$7Akm6Lr0veuUuq2QLHCzpZaHfDjrUe_yK0J-5UFkWPQ?via=matrix.org&via=grusbv.com&via=community.rs

plaes, 2023-11-29:
@Dirbaio Possibly fixed with embassy-time-1.0.5 via https://github.com/embassy-rs/embassy/pull/2074 ?

Dirbaio, 2023-11-29:
yep, thanks!

closed 2023-11-29 by Dirbaio