embassy-rs/embassy #1120: DNS support?

overheat, 2022-12-21:
I found that drogue-device already implement a DNS in `device/src/net/dns.rs`. It's not actually send DNS request (query), but a static IP.

I am not sure. Do we need to implement wrapping of smoltcp's DNS in embassy-net? Then application could leverage it.

lulf, 2022-12-21:
I think the way forward here would be to create a generic async DNS client that implements the DNS traits and uses the UDP traits from embedded-nal-async to do the UDP communication. Embassy-net and others can then implement the traits from e-n-a and it would work other networking implementations as well (like AT-command drivers).

I think this can live in a separate repo as it's not really dependent on embassy and should be possible to implement just using the async traits from e-n-a.

Dirbaio, 2022-12-21:
smoltcp already has a `DnsSocket`. `embassy-net` should get a wrapper for that. 

It's a socket instead of a stack-independent impl on top of UDP because that allows constructing the DNS request and parsing the DNS response in-place inside the network device's buffers. Otherwise if you use `UdpSocket` you need extra buffers to hold the tx/rx queues. There's some discussion on that here https://github.com/smoltcp-rs/smoltcp/pull/465

This does mean you can't use it with an AT-command-based UDP stack. However, these often already have a DNS client (or even allow directly specifying DNS names in commands like TCP connect), so I think it's rather uncommon to need a custom DNS on top of that...

overheat, 2023-01-05:
> 

I tried DNS example in smoltcp, but looks it not work.
https://github.com/smoltcp-rs/smoltcp/issues/731

closed 2023-01-05 by overheat