embassy-rs/embassy #1137: Incorrect delay time with Nucleo-68 (STM32WB55RG)

chrenderle, 2023-01-02:
Hi,

I've learned of embassy today and wanted to experiment a little bit with it on my Nucleo-68 which has a STM32WB55RG. On the blinky example I noticed that the timing was wrong. The time shown via the info! output was ~3 times slower. I've tried several things but nothing worked. It seems like this is a bug in rcc (see matrix discussion [here)](https://matrix.to/#/!YoLPkieCYHGzdjUhOK:matrix.org/$ok87569OEdiJBUuXIeC2Py-5kHvzciMiwqJUftdFe_A?via=matrix.org&via=matrix.ciphernetics.nl). My project which has the issue can be found here: [https://github.com/chrenderle/embassy-nucleo-68-test](https://github.com/chrenderle/embassy-nucleo-68-test).

If someone could point me in the right direction I might be able to fix the bug myself.

lulf, 2023-01-20:
Were you able to resolve this? For RCC, the initialization code for your board is here https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/rcc/wb.rs#L105  - To debug this further, check the RM register configuration with what's done there to make sure clock setup is correct.

chrenderle, 2023-01-20:
> Were you able to resolve this? For RCC, the initialization code for your board is here https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/rcc/wb.rs#L105  - To debug this further, check the RM register configuration with what's done there to make sure clock setup is correct.

Sadly no. The solution of hackwerken didn't work for me (and I forgot to answer to him). 

I'll look into it when I find the time. But I've looked over the code a bit before but couldn't find anything.

chrenderle, 2023-01-21:
Okay, so I looked into the code a bit more and found the issue. At this line [https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/rcc/wb.rs#L67](https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/rcc/wb.rs#L67) it returns 1 for NotDivided. This is the same value as Div3. When looking at the datasheet (RM0434 p. 227) you can see, that the value 0001 in fact represents the divider 3. 
[image]

When changing the line to return 0 it works for me. Since this fixes the issue for me (tested on hardware) I will create a PR for this fix.

closed 2023-01-21 by chrenderle

chrenderle, 2023-01-21:
I thought I'd look over the other processors' code to check if this error is present there too. So I found out that it is also present for the STM32WL processors ([https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/rcc/wl.rs#L161](https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/rcc/wl.rs#L161)). I will add this fix in the upcoming days when I get to it.