embassy-rs/embassy #1146: General question regarding 'ergonomics' of use

mryndzionek, 2023-01-06:
Hello,
I like `embassy` very much, but I have one question regarding on thing that is tedious to me - spelling out long types explicitly. [Here](https://github.com/mryndzionek/max7219-embassy/blob/main/examples/src/bin/example_1.rs#L30)  is an example. When everything is in `main` there is no problem. Compiler can infer the type. As soon as I want to move something to separate function the problems start. It always takes me quite a bit of time to figure out the type signatures. So my questions are:
 - is there maybe some better solution? (using some more generic type in the signature etc.)
 - what is the best way to go about it? Maybe there is some trick/technique that allows for recovering/getting the signatures more easily (rust-analyzer seems to display them in a simplified way)

lulf, 2023-01-06:
I don't think there are a lot of ways around that in Rust, because generics is needed to remove code duplication while still knowing the size of types. However, I think there are some work in embassy specifically like #980 which should simplify type signatures quite a bit. But for drivers that need to work over 'any SPI bus', it's hard to get away with it.

In some cases, it can help using a board support package (like https://github.com/drogue-iot/drogue-device/tree/main/boards/microbit) which can introduce better-named type aliases.

Using type aliases is nice once you know the types, but for knowing the type I usually rely on compiler/rust-analyzer to help me, but I'm not aware of any automated approach.

lulf, 2023-01-20:
Feel free to reopen if you think this is not resolved.

closed 2023-01-20 by lulf