embassy-rs/embassy #1156: `embassy-rp` cannot take TIMER_IRQ_*

tnishinaga, 2023-01-17:
# Question

Hello everyone,

I got `duplicate symbol: TIMER_IRQ_3` error when I used `TIMER_IRQ_3` as the interrupt source for the InterruptExecutor using embassy_rp.

I guess the cause of this problem is that 'TIMER_IRQ_*' are already used in 'embassy-rp/src/timer.rs'.
But, I can't fix it because I don't know how `TIMER_IRQ_*` functions are used.

Could you tell me how `TIMER_IRQ_*` functions are used to fix it?

Thanks

# How to reproduce

1. Create this code in `examples/rp/src/bin/multiprio.rs`.

```
// This code was written with reference to examples/nrf/src/bin/multiprio.rs

#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]

use embassy_executor::{Executor};
use embassy_rp::executor::InterruptExecutor;
use static_cell::StaticCell;
use embassy_rp::interrupt::{self, InterruptExt};
use {defmt_rtt as _, panic_probe as _};
use defmt::unwrap;

static EXECUTOR_TICK: StaticCell<InterruptExecutor<interrupt::TIMER_IRQ_3>> = StaticCell::new();
static EXECUTOR_IDLE: StaticCell<Executor> = StaticCell::new();

#[embassy_executor::task]
async fn time_slice_task() {
    loop {
        todo!();
    }
}

#[embassy_executor::task]
async fn idle_task() {
    loop {
        todo!();
    }
}

#[cortex_m_rt::entry]
fn main() -> ! {
    let _p = embassy_rp::init(Default::default());

    let irq = interrupt::take!(TIMER_IRQ_3);
    irq.set_priority(interrupt::Priority::P3);
    let executor = EXECUTOR_TICK.init(InterruptExecutor::new(irq));
    let spawner = executor.start();
    unwrap!(spawner.spawn(time_slice_task()));

    let executor = EXECUTOR_IDLE.init(Executor::new());
    executor.run(|spawner| {
        unwrap!(spawner.spawn(idle_task()));
    });
}
```

2. move to `examples/rp` and build it

```
cd examples/rp
cargo build
```

# error message

```
~/s/e/e/rp ❯❯❯ cargo build
   Compiling embassy-rp-examples v0.1.0 (/Users/tnishinaga/src/embassy/examples/rp)
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "/var/folders/tq/4zk02pb94hv3zrswytkgz02c0000gn/T/rustcnjObLX/symbols.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.12cwte47wz4x4tub.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.14kt93cg82cal117.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.1i9vvfgjvdrbp8pb.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.1me9r3qje23unxr0.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.1mu07ezgli437bhx.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.1rbu56xm5bb239u8.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.1zyrpxy146vmbj30.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.21v9zxys4bcrwyx4.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.33pi04bbrutpmctw.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.3biw98nkqkhdxvna.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.3osop8d2u69bq34o.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.43j46ff904lnmtxq.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.4e4ado8zrjwtx5w6.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.4vp74dz3897b9pce.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.4ymf82atdlunyfk7.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.5c4ggdkk2nw5ro9t.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.gafg8lv7oq7i998.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.gy96wwqlzdnva9n.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.n6oq10k2pmbmyjp.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.r7d1tpningthdkj.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.uh27ovcsjplt0b1.rcgu.o" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.wdum0zkxou1v1mg.rcgu.o" "--as-needed" "-L" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps" "-L" "/Users/tnishinaga/src/embassy/examples/rp/target/debug/deps" "-L" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/build/embassy-rp-examples-b8f50ed49cdd1baf/out" "-L" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/build/cortex-m-a42fc8b38590c940/out" "-L" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/build/cortex-m-rt-285dde3732cd2ec1/out" "-L" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/build/defmt-6beff97b231ca50e/out" "-L" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/build/embassy-rp-00a08c72b01461ac/out" "-L" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/build/rp2040-pac2-c47d1c03d67498ab/out" "-L" "/Users/tnishinaga/.rustup/toolchains/nightly-2022-11-22-aarch64-apple-darwin/lib/rustlib/thumbv6m-none-eabi/lib" "-Bstatic" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libcortex_m_rt-dd3224841e191168.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libpanic_probe-800b5df8b4e62b05.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libdefmt_rtt-592f2afee79738af.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libembassy_rp-8d05f23581261da3.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/librand_core-c1e21dc619bcce06.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libembedded_hal_nb-a883430a18c7f13f.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libembedded_io-6c40ae5d68fc4194.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/librp2040_pac2-44cd8243a8d61a00.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libembassy_usb_driver-c9d5fe38bda4c63c.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libembassy_futures-f28e0c9dd84c3736.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libembassy_embedded_hal-210c2449ffa9bedf.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libembedded_storage_async-8520f9b605ffe7be.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libembedded_storage-ae79f2db36d67507.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libpio-1f1bbeeb160a906d.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libnum_enum-daf2735ec1bfcb85.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libarrayvec-067f413c2dc793d9.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libembedded_hal_async-b1c24503eb86091e.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libembedded_hal-32f1a3c1a62e8559.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libembassy_cortex_m-2cac9011ea4ee7af.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libcortex_m-5f72843de3caf7f5.rlib" "/Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libvolatile_register-e73d8c70acfe5f39.rlib" "/Users/tnish [... cut ...]
          >>>            /Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/multiprio-0a9665cceeacdba4.gafg8lv7oq7i998.rcgu.o:(TIMER_IRQ_3)
          >>> defined at timer.rs:170 (src/timer.rs:170)
          >>>            embassy_rp-8d05f23581261da3.12a9neuzvynjdv89.rcgu.o:(.text.TIMER_IRQ_3+0x1) in archive /Users/tnishinaga/src/embassy/examples/rp/target/thumbv6m-none-eabi/debug/deps/libembassy_rp-8d05f23581261da3.rlib
          

error: could not compile `embassy-rp-examples` due to previous error
```

lulf, 2023-01-17:
You can disable the 'integrated-timers' feature on the executor, the 'time-driver' feature on embassy-rp, and the 'defmt-timestamp-uptime' feature on embassy-time.

Otherwise the interrupt will be taken by embassy-rp to implement time-driver (which is required for integrated-timers on the executor).

Dirbaio, 2023-01-17:
Keep in mind that if you do that `embassy_time` stops working. 

If you need any random unused interrupt for `InterruptExecutor` then use another one, as `TIMER_IRQ_*` are not unused.

tnishinaga, 2023-01-20:
Thank you very much for your help, lulf-san and dirbaio-san :)

> You can disable the 'integrated-timers' feature on the executor, the 'time-driver' feature on embassy-rp, and the 'defmt-timestamp-uptime' feature on embassy-time.

I was able to take `TIMER_IRQ_*` after disabling the timer-related features.

> If you need any random unused interrupt for InterruptExecutor then use another one, as TIMER_IRQ_* are not unused.

The RP2040 has few interrupt sources, and I couldn't find any interrupt source other than TIMER_IRQ that can be used for periodic interrupts.

> Keep in mind that if you do that embassy_time stops working.

As you pointed out, it is inconvenient that `embassy_time` stops working.
So, I will look for another way.

closed 2023-01-20 by tnishinaga

Dirbaio, 2023-01-20:
@tnishinaga 

Note that `InterruptExecutor` is NOT used for "run this code when the interrupt fires". Use `#[interrupt]` for that, just like in regular non-embassy rust embedded projects.

Anyway, if you want to run something periodically, you can do it in a task:

```rust
#[embassy_executor::task]
async fn tick_task() {
   loop {
      do_thing();
      Timer::after(Duration::from_secs(1)).await;
   } 
}
```

then spawn it at the start of `main`. This will internally use the `TIMER` internally, but through `embassy-time`, so it can be shared with everything else, instead of requiring exclusive access to it like when you use raw interrupts.