embassy-rs/embassy #1157: Support for USB Audio Endpoints

cgudrian, 2023-01-17:
Endpoints for USB Audio devices need two extra bytes: `bRefresh` and `bSynchAddress`.

Dirbaio, 2023-01-17:
They use isochronous endpoints I think, which is also missing.

cgudrian, 2023-01-17:
It depends. Audio streaming uses isochronous or interrupt endpoints, MIDI streaming uses bulk endpoints.

Are there other endpoint descriptor variants?

cgudrian, 2023-01-17:
What about something like this:
```rust
alt.endpoint_bulk_out(64, EndpointExtra::Audio { refresh: 0, synch_address: 0 });
```

nitroxis, 2023-02-01:
I have made a mostly functional USB Audio Class 1.0 implementation for Embassy. However, there are some things missing in embassy-usb that prevent it from being finished: There is currently no way to handle control in/out packets for endpoints (this is required to handle changes to stream parameters such as sample rate or number of channels), and [this issue](https://github.com/embassy-rs/embassy/issues/1176) is kinda blocking third-party USB device crates, as it's not possible to pass around builder structs.

zUnixorn, 2023-08-13:
@nitroxis did you finish the Audio Class implementation? I was looking for a USB Audio Class implementation for embassy but could only find one for the usb-device crate.

TSS-22, 2026-10-01:
Hello,
Is anybody working on the implementation of the USB Audio Class 2.0 for embassy-usb, especially the voice/sound recorder class? I am looking into implementing it (personal need) but if somebody is already on it, I am willing to help and avoid unnecessary effort on what has already been done.

Also is this still the case ?
> They use isochronous endpoints I think, which is also missing support.

Thanks for your time!

leftger, 2026-10-08:
I'm gonna close this issue and Open up a new one concerning USB UAC 2.0.

closed 2026-10-08 by leftger