embassy-rs/embassy #1163: Support for Ramcode

erazor-de, 2023-01-20:
Sometimes one needs his code to be fully executable in RAM. For example when writing the internal FLASH initiated from the firmware. Or like in my case I want to have ramcode which, in combination with a gdb script, can write an external FLASH.

While single functions can be put into ram with
```rust
#[link_section = ".data"]
#[inline(never)]
```
I don't know how to put a whole program into RAM.

The easiest solution I can think of would be to exchange the linker script to put everything into ram and don't handle NVIC entries and use a custom startup function which doesn't need to copy the data section, explicitly set NVIC entries and which is called from external code.

Is it already possible to completely put an embassy project into RAM or is this a feature to be considerate of at all? Is this an effort to be startet here or maybe is it a question to ask in lower crates like cortex-m-rt first?

lulf, 2023-01-20:
Not sure if this is what you're looking for, but the HIL tests for embassy are running in RAM, so maybe look at those for inspiration:

https://github.com/embassy-rs/embassy/tree/master/tests/rp
https://github.com/embassy-rs/embassy/blob/master/tests/rp/link_ram.x

Dirbaio, 2023-03-14:
as @lulf said, this is already possible if you fork the cortex-m-rt linker script. No changes to Embassy are required.

closed 2023-03-14 by Dirbaio