embassy-rs/embassy #1165: Build fails when using a workspace

cgudrian, 2023-01-20:
When building an Embassy project from within a Cargo workspace the build fails with
```
Compiling stm32-metapac v0.1.0 (https://github.com/embassy-rs/embassy.git#825b6710)
LLVM ERROR: Global variable '__INTERRUPTS' has an invalid section specifier '.vector_table.interrupts': mach-o section specifier requires a segment and section separated by a comma.
```
Rust is building the crate `stm32-metapac` for the host instead of the target platform (which can be confirmed by looking at the verbose build output).

When removing the `Cargo.toml` file with the workspace declaration the build succeeds.

Dirbaio, 2023-01-20:
Make sure you have `resolver=2` in the `[workspace]`.

cgudrian, 2023-01-20:
That, of course, does the trick. Thanks for the quick help!

closed 2023-01-20 by cgudrian