embassy-rs/embassy #1167: stm32-metapac: Could not load chip STM32L475VG

milesgranger, 2023-01-21:
Trying to run the example for STM32L475VGT6, after updating `memory.x` and `.cargo/config.toml`'s runner and target. But seems stm32-metapac fails to build:

```
embassy/examples/stm32l4 on  master [!] is 📦 v0.1.0 via 🦀 v1.68.0-nightly took 3s
✦ ❯ cargo run --bin blinky
   Compiling stm32-metapac v0.1.0 (/home/milesg/Projects/embassy/stm32-metapac)
error: failed to run custom build command for `stm32-metapac v0.1.0 (/home/milesg/Projects/embassy/stm32-metapac)`

Caused by:
  process didn't exit successfully: `/home/milesg/Projects/embassy/examples/stm32l4/target/debug/build/stm32-metapac-ddbb7cb15d895643/build-script-build` (exit status: 101)
  --- stdout
  Generating STM32L475VG...

  --- stderr
  thread 'main' panicked at 'Could not load chip STM32L475VG: Os { code: 2, kind: NotFound, message: "No such file or directory" }', /home/milesg/Projects/embassy/stm32-metapac-gen/src/lib.rs:225:40
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: failed to run custom build command for `stm32-metapac v0.1.0 (/home/milesg/Projects/embassy/stm32-metapac)`

Caused by:
  process didn't exit successfully: `/home/milesg/Projects/embassy/examples/stm32l4/target/debug/build/stm32-metapac-91f6f00b024312bb/build-script-build` (exit status: 101)
  --- stdout
  Generating STM32L475VG...

  --- stderr
  thread 'main' panicked at 'Could not load chip STM32L475VG: Os { code: 2, kind: NotFound, message: "No such file or directory" }', /home/milesg/Projects/embassy/stm32-metapac-gen/src/lib.rs:225:40
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Maybe I'm missing something obvious. :sweat_smile:

Dirbaio, 2023-01-21:
Make sure the stm32-data submodule is cloned: `git submodule update --init`

milesgranger, 2023-01-21:
Oh my, indeed. Thank you. 👌

closed 2023-01-21 by milesgranger