embassy-rs/embassy #1175: Is there an ability to export/persist .hex files?

sureshjoshi, 2023-01-25:
I'm using Embassy and the NRF Soft Device repo, with the hopes of doing some basic advertising and authentication, on an NRF 52840.

Where I get a bit stuck is that I'm using `nrjprog` to deploy to my firmware, and shipping binary files isn't reliable. Is there the capability to emit a hex file instead?

lulf, 2023-01-25:
This is the way I've used to generate hex files:

* `cargo install cargo-binutils`
* `rustup component add llvm-tools-preview``
* `cargo objcopy --release -- -O ihex output.hex`

You can use the `mergehex` utility from Nordic to merge this with the softdevice hex to get a single .hex to ship.

sureshjoshi, 2023-01-25:
Thanks, that's perfect!

closed 2023-01-25 by sureshjoshi