embassy-rs/embassy #1181: RP2040 SPI asserts equal length for transfer

elpiel, 2023-01-28:
It seems that the current implementation of Spi requires the read and write to be of equal lengths:

https://github.com/embassy-rs/embassy/blob/7e251a25509a02f9388a8352522e1a279ad857b1/embassy-rp/src/spi.rs#L385-L388

However, in [embedded_hal_async:0.2.0-alpha.0](https://docs.rs/embedded-hal-async/0.2.0-alpha.0/embedded_hal_async/spi/trait.SpiBus.html#tymethod.transfer) it is stated that:

> It is allowed for read and write to have different lengths, even zero length

How should we fix this misalignment?

Dirbaio, 2023-01-29:
Yes, this should be allowed in `embassy-rp`.

elpiel, 2023-02-17:
So we should just remove the assertion?

Dirbaio, 2023-02-17:
No, it's a bit more complicated. I think:

If read is longer, we'd need to write a bunch of dummy zeros to get the SPI to clock more, to do the entire read.
If write is longer , it'd need to either do the extra dummy reads, or (preferably) don't do them and let the rx fifo overflow, and ignore whatever "overflow" error happens, and cleanup the garbage left in the fifo for the next transfer.

elpiel, 2023-03-20:
@Dirbaio would you mind giving me a few more points to guide me how to implement this?

I see that all transfer methods are using this `transfer_inner` method which I should adjust accordingly to what you explained in your comment:
https://github.com/embassy-rs/embassy/blob/b6663a013f8632cefbc6d6ab9a9aa7afbcd1314b/embassy-rp/src/spi.rs#L395-L406

> let the rx fifo overflow, and ignore whatever "overflow" error happens, and cleanup the garbage left in the fifo for the next transfer
How would I allow the FIFO the overflow?

it seems that `dma::read` will only read to the read buffer length:

https://github.com/embassy-rs/embassy/blob/b6663a013f8632cefbc6d6ab9a9aa7afbcd1314b/embassy-rp/src/dma.rs#L46

Dirbaio, 2023-03-20:
> it seems that dma::read will only read to the read buffer length:

yes, but if you write bytes to the SPI without reading, the SPI peripheral keeps clocking bits I think. At least it does in STM32. This is what I meant by "let the RX fifo overflow"

elpiel, 2023-03-20:
So if TX > RX length we can leave the RX overflow by not reading more bytes from the DMA and it differs from the first solution where "it'd need to either do the extra dummy reads". In a nutshell, I don't need to do anything in this case?

And what about RX > TX? I cannot extend the `tx` buffer as it's a `*const` correct?
But I also can't create a new tx buffer like e.g. `let mut tx = [0_u8; to_len];` because `to_len` is not a const value.
How would I write dummy bytes in this case?

Dirbaio, 2023-03-20:
>So if TX > RX length we can leave the RX overflow by not reading more bytes from the DMA and it differs from the first solution where "it'd need to either do the extra dummy reads". In a nutshell, I don't need to do anything in this case?

There's 2 ways:

- Read the wanted bytes, then read dummy bytes.
- Read the wanted bytes, let the rest overflow.

The 1st will work for sure. I think the 2nd one should work as well, and if it does it's the better way, because it'll be faster (less work). It might be a bit tricky because maybe it might leave unread data inthe SPI FIFO which can corrupt the *next* read afterwards, perhaps you have to clear it.

> But I also can't create a new tx buffer like e.g. let mut tx = [0_u8; to_len]; because to_len is not a const value.
> How would I write dummy bytes in this case?

you can read/write dummy bytes by setting `incr_read=false, incr_write=false`. https://dirbaio.github.io/rp2040-pac/rp2040_pac2/dma/regs/struct.CtrlTrig.html#method.incr_write and then pointing it to a 1-byte buffer. This way DMA will repeatedly read/write from the same memory location, instead of requiring an N-bytes buffer.

elpiel, 2023-03-20:
> There's 2 ways:
>
>     * Read the wanted bytes, then read dummy bytes.
>
>     * Read the wanted bytes, let the rest overflow.
>
>
> The 1st will work for sure. I think the 2nd one should work as well, and [...]

What do you mean by clearing?
Do you mean that I should just move the read pointer or re-write the data that was just written to it?

> you can read/write dummy bytes by setting `incr_read=false, incr_write=false`. https://dirbaio.github.io/rp2040-pac/rp2040_pac2/dma/regs/struct.CtrlTrig.html#method.incr_write and then pointing it [...]

If I'm doing read or write I need to set both to `incr_read=false, incr_write=false` or `incr_read=false` for read and `incr_write=false` for write?
I see that there is `write_repeated` function in the `DMA` module but it sets both to `false`.

https://github.com/embassy-rs/embassy/blob/b6663a013f/embassy-rp/src/dma.rs#L78-L95

Dirbaio, 2023-03-20:
>What do you mean by clearing?
>Do you mean that I should just move the read pointer or re-write the data that was just written to it?

the SPI peripheral has a data FIFO. DMA moves bytes from the FIFO to RAM. If you let the FIFO overflow, most of the bytes will be lost but a few will be lingering in the FIFO when the transfer is done. When you do another read, the first bytes popped from the FIFO will be these stale bytes, instead of the current ones.

`write` currently does the same (let the FIFO overflow), so it clears it after it's done, here: https://github.com/embassy-rs/embassy/blob/master/embassy-rp/src/spi.rs#L343-L348

> If I'm doing read or write I need to set both to incr_read=false, incr_write=false or incr_read=false for read and incr_write=false for write?
> I see that there is write_repeated function in the DMA module but it sets both to false.

normal write transfers (read mem -> write reg) do incr_read=true because you want to send several bytes from RAM from teh buffer, and incr_write=false because you want to write them all to the same reg.
normal read transfers (read reg -> write mem) do incr_read=false because you want to read all from the same reg, and incr_write=true because you want to write them all to the buffer one after each other

these "dummy" transfers have both to false, because you want to repeatedly read/write the same byte in mem instead.

> Edit: Only problem with write_repeated I can see is that it always sets the read pointer to 0?!

it sets a pointer pointing to the `dummy` variable, that contains a zero. It'll write N zeros.

commit 9c7b9b78 (2023-03-26): Merge #1288

1288: fix(rp): spi transfer r=elpiel a=elpiel

Fixes #1181
PR #1288 (2023-03-26): fix(rp): spi transfer
Fixes #1181
closed 2023-03-26 by bors[bot]