embassy-rs/embassy #1195: Unable to get Nucelo-H723ZG board to run blinky.

Will-Dale-19, 2023-02-07:
The board's communication LED blinks while the program attempts to run, but after a short while it panics and exits with error code 22. The message I receive is pasted below. Any help to getting atbleast blinky to work would be appreciated, thanks!

cargo run --bin blinky --release
    Finished release [optimized + debuginfo] target(s) in 0.37s
     Running `probe-run --chip STM32H723ZGTx target\thumbv7em-none-eabihf\release\blinky`
(HOST) INFO  flashing program (16 pages / 16.00 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
────────────────────────────────────────────────────────────────────────────────
stack backtrace:
   0: HardFaultTrampoline
      <exception entry>
(HOST) ERROR error occurred during backtrace creation: An error with the usage of the probe occurred

Caused by:
    0: An error specific to a probe type occurred
    1: Command failed with status SwdApFault
               the backtrace may be incomplete.
(HOST) ERROR the program panicked
error: process didn't exit successfully: `probe-run --chip STM32H723ZGTx target\thumbv7em-none-eabihf\release\blinky` (exit code: 22)

gridbox, 2023-02-08:
I had similar behavior when trying to get the blinky example running on a Nucleo-STM32L053. I noticed the chip name was also referenced in features for `embassy-stm32` inside of `Cargo.toml`.  Once I made sure the chip name in both `.cargo/config.toml` and `Config.toml` matched, everything started working.  The docs weren't clear on this so it was just trial and error.

Will-Dale-19, 2023-02-08:
My embassy-stm32 has "stm32h723zg", which should be supported as it it is in the embassy-stm32 cargo.toml, were there any other values you needed to change?

gridbox, 2023-02-10:
> My embassy-stm32 has "stm32h723zg", which should be supported as it it is in the embassy-stm32 cargo.toml, were there any other values you needed to change?

No.  That was the only thing I changed.

MarkHerhold, 2023-03-25:
I have the exact same issue but I'm using a STM31F411RE. My runner line in `config.toml` is `runner = "probe-run --chip STM32F411RETx"`

I am able to flash and debug the example blink project that is provided by ST with the STM32CubeIDE.

Schafwolle, 2024-01-06:
I wanted to run blinky on the NUCLEO-F746ZG  and have a similar Issue. Sadly I am a noob to Rust but I am willing learn and help with testing if you have some suggestions.  

I needed to change the Cargo.toml file and  delete the can.rs and  usart_dma.rs file because of not matching PINs. I am using the "stm32f746zg"  feature flag option. 
I have tested it with Windows and Linux. 

I will now try if an older main works.

I hope that helps to help me fix my Issue to.

LorenzoCucchi, 2024-01-28:
I am experiencing the same problem. I'm using a NUCLEO-F756ZGT and it should be supported by both probe-rs and embassy. 
When running the examples the compilation succeeds and probe-rs start to flash the board, the Erasing and Programming phases work but after the "Finished" message the stlink led (LD4) keeps blinking without ending.

I also tried to flash the board using stlink by converting the elf file to binary, in this case the board is flashed correctly but the program, Blinky in this case doesn't produce effects and all examples don't work.  (The boards has no issues.)

I'm willing to help with testing.

gilroy34, 2024-06-04:
this is all that happens whenever I try to run blinky:
`     Running `probe-rs run --chip STM32H723ZGTx target/thumbv7em-none-eabihf/release/blinky`
      Erasing ✔ [00:00:01] [############################################################] 128.00 KiB/128.00 KiB @ 68.96 KiB/s (eta 0s )
  Programming ✔ [00:00:00] [##############################################################] 28.00 KiB/28.00 KiB @ 58.33 KiB/s (eta 0s )    Finished in 2.345s`

no debugging info or anything like what is shown in the documentation, I then ctrl+c to get back to terminal

MCU: h723zg

`probe-rs list` shows my board connected

any idea

doesnotcompete, 2024-09-08:
I have the same problem on a Nucleo-H723ZG and bisected the issue. For me 3e87dae578969874d354c538d8c59e7be491933d is the first bad commit.

I guess the problem might be in how `memory.x` is processed. The H7 family is the only STM32 that has a build configuration with a custom linker script for the examples.

Simply reverting the commit fixes the issue for me.