embassy-rs/embassy #1198: STM32 G4 RCC implementation is missing PLL

robbym, 2023-02-07:
Like the title suggests, the [file implementing clock configuration](https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/rcc/g4.rs) for G4 devices is missing PLL support (my device in particular is the [STM32G491RE](https://www.st.com/en/microcontrollers-microprocessors/stm32g491re.html)):
```rust
/// Clocks configutation
pub struct Config {
    pub mux: ClockSrc,
    pub ahb_pre: AHBPrescaler,
    pub apb1_pre: APBPrescaler,
    pub apb2_pre: APBPrescaler,
    pub low_power_run: bool,
}
```

Other implementations ([like F4](https://github.com/embassy-rs/embassy/blob/ba18656e94ceffc3fe10a7c47c9b93a1d9d4b2d6/embassy-stm32/src/rcc/f4.rs#L27)) have PLL support and will even calculate the PLL parameters given a source clock, source frequency, and target frequency, which is nice.

I'm fairly new to this project so I don't know if someone is already working on this in particular or on generalizing the notion of PLL configuration, which could _probably_ be done from the STM32 devices I've seen. What should the next steps be?

lulf, 2023-02-07:
Thanks for raising the issue to track this. I'm not aware of anyone working on this, the next steps would be:

* Extend the ClockSrc to support PLL configuration similar to other device classes. In general, the RCC configuration differs between different device classes, so I think it's fine to keep a g4 PLL configuration separate from the f4 configuration, rather than sprinkling the code with `#[cfg(..)]`. 
* Check the RM to ensure that the PLL configuration map to the correct register values.
* Implement and raise PR (profit!) :)

robbym, 2023-02-07:
No problem, thanks for the next steps. I'll take a crack at it!

robbym, 2023-02-07:
@lulf The G4 implementation currently specifies its own enums (e.g. AHBPrescaler, APBPrescaler) whereas other implementations specify it in their rcc_xx.yaml file and have them generated. Should that be done before this? And should that be a separate issue?

lulf, 2023-02-08:
I didn't realize there was a difference there. I think generally the generated PAC types should not be exposed from the HAL, so I think it's best to define the enums in the HAL.

kevswims, 2023-03-01:
@robbym have you made progress on this? Otherwise I can take a crack at it. I'm working on an STM32G473 board and I am pretty familiar with the PLLs on it from some previous rust HAL work on https://github.com/stm32-rs/stm32g4xx-hal/tree/master/src/rcc

robbym, 2023-03-02:
@kevswims I started a few weeks ago but have been swamped with work :( feel free!

cstlaurent, 2023-05-25:
Any work started on that? Might give it a go soon, since I will need it on a project that currently runs on HSE directly and getting slammed 😄

closed 2025-01-10 by Dirbaio