embassy-rs/embassy #1212: Improve efficiency of `embassy::net::Stack.poll()`

MabezDev, 2023-02-12:
`poll` calls smoltcp's `poll_at` method in a hot loop (but yields to the executor each iteration). This seems very inefficient because we could just wait until the `poll_at` instant. It also makes the waker registration for the link_state somewhat redundant too, because it gets polled constantly too.

What's the purpose of using a `poll_fn` loop? Could we make this properly async?

Dirbaio, 2023-02-12:
It shouldn't be a hot loop. Have you actually checked if it is, or is it just from reading the code?

It's creating a timer from poll_at and polling it once: https://github.com/embassy-rs/embassy/blob/master/embassy-net/src/lib.rs#L417

ONly if the timer is already expired, it immediately wakes itself. Otherwise, the timer poll will have registered the waker to be woken at `poll_at` expiration.

MabezDev, 2023-02-12:
In my experience yes, it appears to be a hot loop. I only noticed because I put a print inside the `link_state` driver function and it spewed out thousands of prints :sweat_smile:.

Dirbaio, 2023-02-12:
I just double-checked running the nrf `usb_ethernet` example, and it doesn't poll in a hot loop. It might be an issue in the driver.

MabezDev, 2023-02-12:
Yep, you were right! I debugged a bit further and my transmit waker was outside the check for whether we have to actually send something or not... Oops :sweat_smile:. Apologies for the noise.

closed 2023-02-12 by MabezDev

Dirbaio, 2023-02-12:
btw you have to wake the waker only when an earlier  `transmt/receive` call returned None because there was no buffer space, and then buffer space becomes available. No need to wake on all "tx done".

MabezDev, 2023-02-13:
Thanks! I'll look into doing that too :).