embassy-rs/embassy #1214: embassy-net: Separate device from stack?

MabezDev, 2023-02-13:
Right now it seems impossible to add a way to access the underlying device with embassy-net. In its current form it can't simply be added with a `device_mut` method on the stack because the device will be borrowed across await points causing lifetime errors.

It seems to me they need to be split; this is the way smoltcp went anyways, the iface only needs the device and sockets when `poll`ing. I'm happy to work on this but I'd need a bit of direction as the changes needing to be made. My end goal is the have a task like this:

```rust
#[embassy_executor::task]
async fn connection(_stack: &'static Stack<WifiDevice>) {
    println!("start connection task");
    let mut device = unsafe { WifiDevice::new() }; // TODO THIS IS BAD - but there is no way to get access to the device in the stack at the moment
    println!("Device capabilities: {:?}", device.get_capabilities());
    loop {
        match esp_wifi::wifi::get_wifi_state() {
            WifiState::StaConnected => {
                // wait until we're no longer connected
                WifiEvent::StaDisconnected.await;
                Timer::after(Duration::from_millis(5000)).await
            },
            _ => {}
        }
        if !matches!(device.is_started(), Ok(true)) {
            let client_config = Configuration::Client(ClientConfiguration {
                ssid: SSID.into(),
                password: PASSWORD.into(),
                ..Default::default()
            });
            device.set_configuration(&client_config).unwrap();
            println!("Starting wifi");
            device.start().await.unwrap();
            println!("Wifi started!");
        }
        println!("About to connect...");
        
        match device.connect().await {
            Ok(_) => println!("Wifi connected!"),
            Err(e) => { 
                println!("Failed to connect to wifi: {e:?}");
                Timer::after(Duration::from_millis(5000)).await
            },
        }
    }
}
```
where I _don't_ have to conjure the device to handle the uplink process.

Perhaps I'm going about this the wrong way, please feel free to tell me so! :)

Dirbaio, 2023-02-13:
I'd rather not do this. It increases the API complexity quite a bit, requiring the user to write the poll loop themselves. Exposes implementation details and gives the user too much rope to hang themselves with. I regularly see people hanging themselves in the smoltcp issue tracker :)

(smoltcp has to do it that way because it has to stay fully unopinionated. embassy-net can be more opinionated because it can assume you have a multitasking async executor).

Even if embassy-net requires `&mut` on the device, it doesn't mean the device internally can't share state via mutexes/atomics. You can return two objects: the device, and a "control" object that can be used to join a wifi network, query state, etc. 

This is how cyw43 does it: https://github.com/embassy-rs/cyw43/blob/master/examples/rpi-pico-w/src/main.rs#L69

MabezDev, 2023-02-13:
Thanks, I actually just stumbled across the cyw43 repo right after posting this :D. I like the control struct idea, and luckily the control and packet-sending parts of the driver are independent anyways, hence why creating a driver unsafely in the task above works.

closed 2023-02-13 by MabezDev