embassy-rs/embassy #1223: embassy-net: TCP sockets discards unset data on Drop

kbleeke, 2023-02-21:
I have a relatively simple program that
1. accepts a new connection
2. writes some bme280 measurements to the socket
3. drops the socket and waits for a new connection

This results in no data being transmitted. As far as I can tell, the problem is that Drop immediately unregisters the socket's buffers from the network stack so that the stack never gets to execute the write.

I would have expected all outdoing data to be sent. Alternatively, there should be a method that waits for this.

Dirbaio, 2023-02-21:
This is intended behavior. It *has* to be this way because the data is in the buffers you provide. If you drop the socket and deallocate the buffers, the stack has nowhere to store it.

One solution would be to make `.flush()` wait until all the data is ack'd at the TCP level. It's not currently implemented, but PR's welcome.

However, in practice, if you want this to be reliable, you probably want some "application-level ACK" that the data was received correctly on the other end. Waiting for TCP ACKs is not enough, because:
- ACK only means it has reached the remote OS's TCP stack, it doesn't mean the data has been delivered to the application and the application has successfully stored it on a database or whatever.
- If you go through WAN, there can be TCP middleboxes / proxies that will ACK the data and then take charge of sending it to the peer.

kbleeke, 2023-02-22:
That makes sense. Thanks for the tip for application-level ACKs. I don't think, I require them specifically in my case but waiting for incoming data would certainly fix my issue.

I think I came up with an implementation for flush, I've opened a PR

commit 464faa2a (2023-02-22): Merge #1226

1226: embassy-net: Implement flush for TcpSocket r=Dirbaio a=kbleeke

Implements flush for TcpSocket by checking the send queue. 

Flushing is implemented by checking if smoltcp's send_queue/tx_buffer is empty. The flush is completed when all outstanding octets are acknowledged. Smoltcp wakes the send waker [here](https://docs.rs/smoltcp/latest/src/smoltcp/socket/tcp.rs.html#1712) when ACKs are processed and data is removed from the send buffer. So we can re-check in our flush implementation, if the buffer is now empty.

fixes #1223
PR #1226 (2023-02-22): embassy-net: Implement flush for TcpSocket
Implements flush for TcpSocket by checking the send queue. 

Flushing is implemented by checking if smoltcp's send_queue/tx_buffer is empty. The flush is completed when all outstanding octets are acknowledged. Smoltcp wakes the send waker [here](https://docs.rs/smoltcp/latest/src/smoltcp/socket/tcp.rs.html#1712) when ACKs are processed and data is removed from the send buffer. So we can re-check in our flush implementation, if the buffer is now empty.

fixes #1223
closed 2023-02-22 by bors[bot]