embassy-rs/embassy #1229: embassy-net-driver-channel: large stack allocation for `State`

LimpidCrypto, 2023-02-23:
I need a `RUST_MIN_STACK` of approx. 90000000 (90 MB) to get my test running. The critical part seems to be the initiation of a new `embassy_net_driver_channel::State`. Could someone please explain to me why I need to define `RUST_MIN_STACK` to be that large? I would love to know what I'm doing wrong.

```rust
trait Sizes {
    const MTU: usize;
    const BUF: usize;
}

pub struct TcpStack {
    pub inner: embassy_net::Stack<
        embassy_net_driver_channel::Device<'static, { <TcpStack as Sizes>::MTU }>,
    >,
}

impl Sizes for TcpStack {
    const MTU: usize = ETHERNET_MTU + ETHERNET_HEADER_LEN; // 1514
    const BUF: usize = TCP_BUF; // 4096
}

impl TcpStack {
    pub fn new(mac_address: [u8; 6]) -> Self {
        let state = singleton!(embassy_net_driver_channel::State::<
            { <TcpStack as Sizes>::MTU },
            { <TcpStack as Sizes>::BUF },
            { <TcpStack as Sizes>::BUF },
        >::new());
        let device = StackDevice::new(mac_address, state);
        let config = embassy_net::ConfigStrategy::Dhcp;
        let resources: &mut embassy_net::StackResources<1, 2, 8> =
            singleton!(embassy_net::StackResources::new());

        let mut seed = [0; 8];
        rand::thread_rng().fill_bytes(&mut seed);
        let seed = u64::from_le_bytes(seed);

        let stack = embassy_net::Stack::new(device.inner, config, resources, seed);

        Self { inner: stack }
    }
}

#[cfg(test)]
mod test {
    use super::*;

    use anyhow::Result;

    #[test]
    fn test_new() -> Result<()> {
        let stack = TcpStack::new([0xa4, 83, 0xe7, 48, 31, 21]);

        Ok(())
    }
}
```

`StackDevice` is also just a wrapper with an `inner` which is `embassy_net_driver_channel::Device`

Dirbaio, 2023-02-23:
the 2nd and 3rd generic args to State are the *count of MTU-sized buffers*, not the buffer size. You're allocating 1514*4096*2 = 12MB there. 

Set them to something small, like `4`.

LimpidCrypto, 2023-02-23:
> the 2nd and 3rd generic args to State are the _count of MTU-sized buffers_, not a buffer size in bytes. You're allocating `1514*4096*2 = 12MB` there.
> 
> Set them to something small, like `4`.

Ah great, thank you 🙏

closed 2023-02-23 by LimpidCrypto