embassy-rs/embassy #1233: Cannot run TCP Stack on latest nightly

LimpidCrypto, 2023-02-23:
I'm trying to run the following test with `--feature "test"` so it's testing with std (if this could be relevant for my issue):
```rust
#[cfg(test)]
#[cfg(feature = "test")]
mod test {
    use super::*;
    use embassy_net::Ipv4Address;

    #[tokio::test]
    async fn test_connect() {
        let mac_address = [0xa4, 0x83, 0xe7, 0x48, 0x31, 0x21];
        let ip = Ipv4Address::new(92, 204, 239, 18);
        let remote = (ip, 80);
        let stream = TcpStream::new(mac_address, remote);

        stream.connect().await;
    }
}
```
The `TcpStream` currently looks like this:
```rust
pub struct TcpSocket {
    pub remote: (Ipv4Address, u16),
    pub socket: RefCell<embassy_net::tcp::TcpSocket<'static>>,
    pub stack: &'static TcpStack,
}

impl TcpSocket {
    pub fn new(mac_address: [u8; 6], remote: (Ipv4Address, u16)) -> Self {
        let stack: &'static TcpStack = &*singleton!(TcpStack::new(mac_address));

        let rx_buffer: &'static mut [u8; TCP_BUF] = singleton!([0; TCP_BUF]);
        let tx_buffer: &'static mut [u8; TCP_BUF] = singleton!([0; TCP_BUF]);

        let socket = embassy_net::tcp::TcpSocket::new(&stack.inner, rx_buffer, tx_buffer);

        Self {
            remote,
            socket: RefCell::new(socket),
            stack,
        }
    }

    pub async fn connect(&self) {
        let pair = embassy_futures::join::join(self.stack._run(), self._do_connect()).await;
    }

    async fn _do_connect(&self) -> Result<(), ConnectError> {
        self.socket
            .borrow_mut()
            .set_timeout(Some(embassy_net::SmolDuration::from_secs(10)));
        self.socket.borrow_mut().connect(self.remote).await
    }
}
```
**But the problem actually occurs as soon as **`TcpSocket.connect` calls `embassy_futures::join::join(self.stack._run(), ...)`**. You find `TcpStack._run()` here:
```rust
trait Sizes {
    const MTU: usize;
    const MTU_COUNT: usize;
}

pub struct TcpStack {
    pub inner: embassy_net::Stack<
        embassy_net_driver_channel::Device<'static, { <TcpStack as Sizes>::MTU }>,
    >,
}

impl Sizes for TcpStack {
    const MTU: usize = ETHERNET_MTU + ETHERNET_HEADER_LEN; // 1514
    const MTU_COUNT: usize = ETHERNET_MTU_COUNT;
}

impl TcpStack {
    pub fn new(nic_mac_address: [u8; 6]) -> Self {
        let state = singleton!(embassy_net_driver_channel::State::<
            { <TcpStack as Sizes>::MTU },
            { <TcpStack as Sizes>::MTU_COUNT },
            { <TcpStack as Sizes>::MTU_COUNT },
        >::new());
        let device = Nic::new(nic_mac_address, state);
        let config = embassy_net::ConfigStrategy::Dhcp;
        let resources: &mut embassy_net::StackResources<1, 2, 8> =
            singleton!(embassy_net::StackResources::new());

        let mut seed = [0; 8];
        rand::thread_rng().fill_bytes(&mut seed);
        let seed = u64::from_le_bytes(seed);

        let stack = embassy_net::Stack::new(device.inner, config, resources, seed);

        Self { inner: stack }
    }

    /*************************
     * This causes the issue *
     *************************/
    // Simply runs the `embassy_net::Stack`
    pub async fn _run(&self) {
        self.inner.run().await
    }
}
```
The error that's being thrown while compiling is the following:
```
Undefined symbols for architecture x86_64:
    "__embassy_time_now", referenced from:
        embassy_time::driver::now::h1e0e006f20ccee30 in libembassy_time-d5b6b12840936136.rlib(embassy_time-d5b6b12840936136.embassy_time.c50af4c3-cgu.0.rcgu.o)
    "__embassy_time_schedule_wake", referenced from:
        embassy_time::timer::schedule_wake::hc01de17fa5ee189b in libembassy_time-d5b6b12840936136.rlib(embassy_time-d5b6b12840936136.embassy_time.c50af4c3-cgu.3.rcgu.o)
  ld: symbol(s) not found for architecture x86_64
  clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
I'm a little bit unsure if this is an issue of this library. It would be awesome someone could point me into the right direction how I could fix this issue, thank you :)

Dirbaio, 2023-02-23:
Check that  `embassy-time` has the `std` Cargo feature enabled.

If you want to enable it just for tests, you can put it in `[dev-dependencies]`

LimpidCrypto, 2023-02-23:
I don't use the `embassy-time` crate. Here is my `Cargo.toml`:
```
[features]
std = [
    "embassy-net/std",
    "tokio/full",
]
test = [
    "embassy-net/std",
    "tokio/full",
]

[dependencies]
... all non-embassy

[dependencies.embassy-net]
git = "https://github.com/embassy-rs/embassy"
package = "embassy-net"
version = "0.1.0"
features = ["nightly", "tcp", "medium-ethernet", "dhcpv4", "unstable-traits"]

[dependencies.embassy-net-driver-channel]
git = "https://github.com/embassy-rs/embassy"
package = "embassy-net-driver-channel"
version = "0.1.0"

[dependencies.embassy-futures]
git = "https://github.com/embassy-rs/embassy"
package = "embassy-futures"
version = "0.1.0"
```

Dirbaio, 2023-02-23:
`embassy-net` uses `embassy-time` internally, so it requires a time driver configured, even if you're not using it directly.

LimpidCrypto, 2023-02-23:
> `embassy-net` uses `embassy-time` internally, so it requires a time driver configured, even if you're not using it directly.

Alright, I've added it as you suggested, thank you 🙏 
I've also added the `embassy-executor` crate and enabled the `generic-queue` feature for `embassy-time`. So my final `Cargo.toml` looks like this:
```toml
[features]
std = [
    "embassy-executor/std",
    "embassy-net/std",
    "embassy-time/std",
    "embassy-time/generic-queue",
]
test = [
    "embassy-executor/std",
    "embassy-net/std",
    "embassy-time/std",
    "embassy-time/generic-queue",
]

[dependencies]
# ...

[dependencies.embassy-net]
git = "https://github.com/embassy-rs/embassy"
package = "embassy-net"
version = "0.1.0"
features = ["nightly", "tcp", "medium-ethernet", "dhcpv4", "unstable-traits"]

[dependencies.embassy-net-driver-channel]
git = "https://github.com/embassy-rs/embassy"
package = "embassy-net-driver-channel"
version = "0.1.0"

[dependencies.embassy-executor]
git = "https://github.com/embassy-rs/embassy"
package = "embassy-executor"
version = "0.1.1"
features = ["nightly"]

[dependencies.embassy-futures]
git = "https://github.com/embassy-rs/embassy"
package = "embassy-futures"
version = "0.1.0"

[dependencies.embassy-time]
git = "https://github.com/embassy-rs/embassy"
package = "embassy-time"
version = "0.1.0"
```

closed 2023-02-23 by LimpidCrypto