embassy-rs/embassy #1234: embassy-net: NoRoute error on connect

LimpidCrypto, 2023-02-24:
## Source
https://github.com/LimpidCrypto/em-as-net/blob/main/src/core/tcp/mod.rs#L56

## Overview
I'm getting a `ConnectError::NoRoute` error when trying to connect. The error get thrown when `smoltcp` [`get_source_address`](https://github.com/smoltcp-rs/smoltcp/blob/master/src/iface/interface/mod.rs#L1079) returns `None`, when trying to choose a local address automatically. This results in an `smoltcp::tcp::ConnectError::Unaddressable` error. I can't really get my head around what I'm doing wrong. Can someone please help me out here and point me in the right direction what could cause this error?

## Notes
- I can ping the ip
- I'm using the default settings for DHCP

LimpidCrypto, 2023-02-25:
Seems like the default config of DHCP is not working for me. Using static config works doesn't throw an error anymore.
```rust
// `TcpConfig` is just a wrapper for `embassy_net::Config`
let config = TcpConfig::new_static(
    Ipv4Cidr::new(Ipv4Address::new(192, 168, 69, 2), 24),
    Some(Ipv4Address::new(192, 168, 69, 1)),
    Vec::from_iter([
        Ipv4Address::new(8, 8, 8, 8),
        Ipv4Address::new(9, 9, 9, 9),
        Ipv4Address::new(1, 1, 1, 1),
   ]),
);
```
But now connecting to the remote ip takes forever. Any advices?

kbleeke, 2023-02-28:
Are you waiting for DHCP to be finished? I have something like this before I attempt do use TCP

```rust
while !stack.is_config_up() {
    yield_now().await;
}
```

LimpidCrypto, 2023-02-28:
> Are you waiting for DHCP to be finished? I have something like this before I attempt do use TCP
> 
> 
> 
> ```rust
> 
> while !stack.is_config_up() {
> 
>     yield_now().await;
> 
> }
> 
> ```

Ah nice, I'm not doing that currently. I will give it a try later, thank you @kbleeke

LimpidCrypto, 2023-02-28:
Unfortunately it doesn't resolve my problem. It also turned out, that not `TcpSocket.connect` but `Stack.run` takes forever to complete.

Dirbaio, 2023-02-28:
`run` runs forever, you have to launch it in a background task. Please check how the examples do it.

Also, for DHCP to work, you have to bridge instead of route the tap interface. Tun doesn't work. Doing it over Wifi doesn't work. See https://github.com/smoltcp-rs/smoltcp/#bridged-connection

Check whether embassy-net is logging "Acquired IP configuration" after a bit. If it's not, then DHCP is not working, and there's something wrong with your tap interface.

LimpidCrypto, 2023-03-01:
Alright, found some more issues and misconceptions in my code. I will refactor my code.
Thank you both for your input :)

closed 2023-03-01 by LimpidCrypto