embassy-rs/embassy #1235: Bumping embedded-storage-async to 0.4 in embassy_nrf

mehmetalianil, 2023-02-24:
If the Softdevice is enabled, it is required to take the Flash peripheral from the softdevice with: 
```rust
    let mut f = Flash::take(sd);
```
which returns Flash, which implements `ReadNorFlash`,`AsyncReadNorFlash`and`AsyncNorFlash`
the firmware updater for the update sequence is created like this:
```rust
let mut updater = FirmwareUpdater::default();
```
which, when the methods of it are called like this: 
```rust
updater.mark_updated(&mut f, &mut magic).await.unwrap();
//...
updater.write_firmware(updateoffset as usize, &updatebuffer.0, &mut f, 4096).await.unwrap();
//...
```
These functions require f to have the trait `F: AsyncNorFlash` from embedded_storage_async of version 0.3

embedded_storage_async will be updated to 0.4 and all issues will be fixed, probably with nvmc and qspi.

mehmetalianil, 2023-03-06:
embassy_boots also include the embedded-storage-async crate. 
I don't have an RP and STM32 to test these with. Any ideas?

Dirbaio, 2023-03-06:
just make the changes so that it compiles, don't worry about testing. I'll take a look in PR review! Thanks :)

closed 2023-03-14 by Dirbaio