embassy-rs/embassy #1241: embassy-rp Ability to change UART baudrate during runtime

andres-hurtado-lopez, 2023-02-26:
As far as I have worked with Embassy, is of my understanding that  Peripherals and UART interfaces are built under the concept of the UART interface consuming the interface pins from the the Peripherals singleton. 

In rust if a member of a struct is consumed there is no way back. Which means as soon as the UART is created you can't create another instance with a different baudrate. This is not the case of more more high level frameworks, _(sorry for the example I'm going to give here I which I understand is a totally different tool)_ the arduino **Processing** environment on which to change the speed is only a matter of creating again a new UART instance with a new speed configuration.

There is a case in which there are serial devices that work with different buadrates during runtime _(for intance configure the device at 9600 8-N-1 and then go 115200 for regular operation after configuration)_. On the `embassy_rp::uart::Uart` methods there isn't any kind of aid to change the baudrate neither any other configuration of the UART during runtime. I'm missing something on how I'm thinking on how to solve the problem or this is a feature that is missing on the UART functionality ?

Dirbaio, 2023-02-26:
- You can create the uart driver with `&mut` on the peripheral and pins singleton, which means that when the driver is dropped the borrow on the singletons is released, and you can use them again.
- There's no "set baudrate" method currently, but we could have it. [nRF already has it.](https://github.com/embassy-rs/embassy/blob/bc71230cd07296468f2e03c00f9ceddbab67c9d9/embassy-nrf/src/buffered_uarte.rs#LL190C12-L190C24). PR's welcome!.

andres-hurtado-lopez, 2023-02-26:
> * You can create the uart driver with `&mut` on the peripheral and pins singleton, which means that when the driver is dropped the borrow on the singletons is released, and you can use them again.
> * [...]

Alright I have just created the `set_baudrate`:

```
/// sets baudrate on runtime    
pub fn set_baudrate(&mut self, baudrate: u32) {
	
	let r = T::regs();
	
        let clk_base = crate::clocks::clk_peri_freq();
	
        let baud_rate_div = (8 * clk_base) / baudrate;
        let mut baud_ibrd = baud_rate_div >> 7;
        let mut baud_fbrd = ((baud_rate_div & 0x7f) + 1) / 2;
	
        if baud_ibrd == 0 {
            baud_ibrd = 1;
            baud_fbrd = 0;
        } else if baud_ibrd >= 65535 {
            baud_ibrd = 65535;
            baud_fbrd = 0;
        }
	
        unsafe {
	    
	    // Load PL011's baud divisor registers
            r.uartibrd().write_value(pac::uart::regs::Uartibrd(baud_ibrd));
            r.uartfbrd().write_value(pac::uart::regs::Uartfbrd(baud_fbrd));
	}
	
    }
```

 Which is the procedure for the PR ? I need to create a new branch with some specific naming ? What else needs to be submitted to fulfill all the requirements ?

Dirbaio, 2023-02-26:
Standard github procedure (fork repo, push change to a branch in your fork with whatever name you want, open a PR)

Dirbaio, 2023-02-26:
Please don't duplicate the baudrate calculation code. Either extract it to a separate function, or make `new` call `set_baudrate`. I think the latter is nicer if it works.

andres-hurtado-lopez, 2023-02-26:
Please let me know I I understood correctly it would be first some private abstraction for the baudrate calculation like this withthin UART's implementation:

```
    fn baudrate_calculations(baudrate: u32) -> (u32, u32) {
	
        let clk_base = crate::clocks::clk_peri_freq();
	
        let baud_rate_div = (8 * clk_base) / baudrate;
        let mut baud_ibrd = baud_rate_div >> 7;
        let mut baud_fbrd = ((baud_rate_div & 0x7f) + 1) / 2;

        if baud_ibrd == 0 {
            baud_ibrd = 1;
            baud_fbrd = 0;
        } else if baud_ibrd >= 65535 {
            baud_ibrd = 65535;
            baud_fbrd = 0;
        }

	(baud_ibrd, baud_fbrd)
    }

```

and then on either places where is the dependency then do the following calls right ? first on `init`

```
    fn init(
        tx: Option<PeripheralRef<'_, AnyPin>>,
        rx: Option<PeripheralRef<'_, AnyPin>>,
        rts: Option<PeripheralRef<'_, AnyPin>>,
        cts: Option<PeripheralRef<'_, AnyPin>>,
        config: Config,
    ) {
        let r = T::regs();
        unsafe {
            if let Some(pin) = &tx {
                pin.io().ctrl().write(|w| w.set_funcsel(2));
                pin.pad_ctrl().write(|w| w.set_ie(true));
            }
            if let Some(pin) = &rx {
                pin.io().ctrl().write(|w| w.set_funcsel(2));
                pin.pad_ctrl().write(|w| w.set_ie(true));
            }
            if let Some(pin) = &cts {
                pin.io().ctrl().write(|w| w.set_funcsel(2));
                pin.pad_ctrl().write(|w| w.set_ie(true));
            }
            if let Some(pin) = &rts {
                pin.io().ctrl().write(|w| w.set_funcsel(2));
                pin.pad_ctrl().write(|w| w.set_ie(true));
            }

	    let (baud_ibrd, baud_fbrd) = Uart::<T,M>::baudrate_calculations(config.baudrate);

            // Load PL011's baud divisor registers
            r.uartibrd().write_value(pac::uart::regs::Uartibrd(baud_ibrd));
            r.uartfbrd().write_value(pac::uart::regs::Uartfbrd(baud_fbrd));

......

```

and the over the new method that I'm suggesting:

```
    pub fn set_baudrate(&mut self, baudrate: u32) {
	
	let r = T::regs();
	
	let (baud_ibrd, baud_fbrd) = Self::baudrate_calculations(baudrate);

        unsafe {
	    
	    // Load PL011's baud divisor registers
            r.uartibrd().write_value(pac::uart::regs::Uartibrd(baud_ibrd));
            r.uartfbrd().write_value(pac::uart::regs::Uartfbrd(baud_fbrd));
	}
	
    }
```

Is this correct ?

Dirbaio, 2023-02-26:
That works, but I think it should be possible to do something like this, so you don't duplicate the register writes either

```rust
fn init(...) {
    ...
    Self::set_baudrate_inner(baudrate);
    ...
}

/// sets baudrate on runtime    
pub fn set_baudrate(&mut self, baudrate: u32) {
	    Self::set_baudrate_inner(baudrate);
}

fn set_baudrate_inner(baudrate: u32) {
	let r = T::regs();
	
        let clk_base = crate::clocks::clk_peri_freq();
	
        let baud_rate_div = (8 * clk_base) / baudrate;
        let mut baud_ibrd = baud_rate_div >> 7;
        let mut baud_fbrd = ((baud_rate_div & 0x7f) + 1) / 2;
	
        if baud_ibrd == 0 {
            baud_ibrd = 1;
            baud_fbrd = 0;
        } else if baud_ibrd >= 65535 {
            baud_ibrd = 65535;
            baud_fbrd = 0;
        }
	
        unsafe {
	    
	    // Load PL011's baud divisor registers
            r.uartibrd().write_value(pac::uart::regs::Uartibrd(baud_ibrd));
            r.uartfbrd().write_value(pac::uart::regs::Uartfbrd(baud_fbrd));
	}
}
```

Dirbaio, 2023-03-01:
fixed in #1243

closed 2023-03-01 by Dirbaio