embassy-rs/embassy #1272: Embassy-sync: pipe - only partial bytearray written

StefanB7, 2023-03-10:
Weird issue I encountered this morning, can be that I'm not using the pipe correctly but here it is nonetheless:

In essence, if I write u8 byte arrays to the pipe, even when there is space available in the pipe, the `try_write` function does not write the entire array to the pipe if the total number of bytes that have been written to the pipe in total nears the pipe's size.

For example:
I've got a pipe:
```
static SHARED_RX_PIPE: Pipe<ThreadModeRawMutex, 30> = Pipe::new();
```
say we have a byte array u8, of 20 elements we want to write to the pipe, we can do this as such:
```
let array: [u8;20] = [0;20]
let pipe_writer = SHARED_RX_PIPE.writer();
pipe_writer.try_write(&array[..]).unwrap();
```
This works fine, another task now reads all of the values, and the length of the pipe is thus 0 again.

Now if we try to write the same array again, only 10 bytes will be written to the pipe, even if the pipe is empty:
```
pipe_writer.try_write(&array[..]).unwrap(); // No error is returned, but only 10 bytes are written
```

If I loop over each one of the bytes individually and add them (thus not trying to add them as a slice), this works fine, each time.
```
for i in 0..20 {
    pipe_writer.try_write(&[array[i]]).unwrap();
    //pipe_writer_responses.try_write(&[buffer[i]]).unwrap();
}
```
Is there perhaps some wrap-over error occurring?

Thanks for this awesome project. It has made embedded fun again.

Dirbaio, 2023-03-10:
This is sort of working as intended. `Pipe` is a byte stream, so reading/writing work like the `std::io` traits:: they're always allowed to do "short writes". There's no guarantee like "if there is enough space for the entire buffer, then it will be entirely written". 

If you want to write the entire buffer, you have to loop, or you can use [write_all](https://docs.rs/embedded-io/0.4.0/embedded_io/asynch/trait.Write.html#method.write_all) which does it for you.

StefanB7, 2023-03-11:
Thanks for the clarification Dirbaio. I'll close this issue.

closed 2023-03-11 by StefanB7