embassy-rs/embassy #1275: Unable to run blinky on nrf52840 (Seed XIAO Sense) - CPU resets after await

yuriykulikov, 2023-03-12:
I am trying to get https://github.com/embassy-rs/embassy/blob/master/examples/nrf52840/src/bin/blinky.rs running on a https://www.seeedstudio.com/Seeed-XIAO-BLE-nRF52840-p-5201.html

These boards are based on nRF52840 and come shipped with an Adafruit bootloader.

After modifications to `memory.x` and `blinky.rs` I have managed to use LEDs, **but CPU resets after the first `await`**.

Here is my memory.x:

```
MEMORY
{
  FLASH : ORIGIN = 0x27000, LENGTH = 868K
  RAM : ORIGIN = 0x20000000, LENGTH = 256K
}
```

And blinky:

```
#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let p = embassy_nrf::init(Default::default());
    let mut led_r = Output::new(p.P0_26, Level::High, OutputDrive::Standard);
    let mut led_g = Output::new(p.P0_30, Level::High, OutputDrive::Standard);
    let mut led_b = Output::new(p.P0_06, Level::High, OutputDrive::Standard);

    loop {
        led_r.set_low();
        led_g.set_low();
        Timer::after(Duration::from_millis(5000)).await;
        // never see the blue light, board resets after await returns
        led_b.set_low();
        Timer::after(Duration::from_millis(300)).await;
    }
}
```

Unfortunately I do not have a debugger. Can somebody point me in the right direction?

Dirbaio, 2023-03-12:
If the flash contains the softdevice (I guess it does since you made flash start at `0x27000`) then you also have to reserve the first 8 bytes of RAM, even if you're not using the softdevice at all. [Source](https://infocenter.nordicsemi.com/topic/sds_s140/SDS/s1xx/mem_usage/mem_resource_reqs.html?cp=5_7_4_0_13_0_0)

so try this:

```rust
  RAM : ORIGIN = 0x20000008, LENGTH = 0x3fff8
```

yuriykulikov, 2023-03-12:
Thank you @Dirbaio!

This did the trick :)

closed 2023-03-12 by yuriykulikov

michaelrommel, 2023-10-16:
Hello there,

is it possible to add this to the documentation? This took me like 6 hours to find... I think it would be good to have this kind of information somewhere in a board-specific "quirks" document or so. Or have I just been unable to find this despite a lot of effort on my side?

Thanks in advance!!

  Michael.

michaelrommel, 2023-10-17:
I have now opened up a new issue for the documentation and the provided `memory.x` file, that contains the conflicting information: https://github.com/embassy-rs/embassy/issues/2079