embassy-rs/embassy #1280: QSPI example fails on nrf52840, possibly not erasing properly

mehmetalianil, 2023-03-15:
We realized that q.erase() was failing. (tbh I must have seen that coming since erasing each page became faster than chip erase.)
We were able to read data that precedes erase cmd after the erasure.

Therefore I tried the example, since I remember it goes through all of the hoops.

Log:
```
(HOST) INFO  flashing program (16 pages / 64.00 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
0.000183 INFO  id: [194, 32, 25]
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:39
0.000701 INFO  status: 66
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:45
0.001098 INFO  page 0: erasing... 
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:60
0.016571 INFO  programming...
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:67
0.020965 INFO  page 1: erasing... 
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:60
0.036468 INFO  programming...
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:67
0.040924 INFO  page 2: erasing... 
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:60
0.056396 INFO  programming...
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:67
0.060821 INFO  page 3: erasing... 
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:60
0.076293 INFO  programming...
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:67
0.080718 INFO  page 4: erasing... 
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:60
0.096252 INFO  programming...
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:67
0.100708 INFO  page 5: erasing... 
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:60
0.116241 INFO  programming...
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:67
0.120635 INFO  page 6: erasing... 
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:60
0.136169 INFO  programming...
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:67
0.140625 INFO  page 7: erasing... 
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:60
0.156127 INFO  programming...
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:67
0.160552 INFO  page 0: reading... 
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:72
0.161163 INFO  verifying...
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:75
0.161376 ERROR panicked at 'assertion failed: `(left == right)`'
diff < left / right >
<136
>0
└─ qspi::____embassy_main_task::{async_fn#0} @ src\bin\qspi.rs:77
0.161682 ERROR panicked at 'explicit panic', C:\Users\Mehmet Ali Anil\.cargo\registry\src\github.com-1ecc6299db9ec823\defmt-0.3.2\src\lib.rs:367:5
└─ panic_probe::print_defmt::print @ C:\Users\Mehmet Ali Anil\.cargo\registry\src\github.com-1ecc6299db9ec823\panic-probe-0.3.0\src\lib.rs:91
────────────────────────────────────────────────────────────────────────────────
stack backtrace:
   0: HardFaultTrampoline
      <exception entry>
   1: lib::inline::__udf
        at ./asm/inline.rs:181:5
   2: __udf
        at ./asm/lib.rs:51:17
   3: cortex_m::asm::udf
        at C:\Users\Mehmet Ali Anil\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-0.7.7\src\asm.rs:43:5
   4: rust_begin_unwind
        at C:\Users\Mehmet Ali Anil\.cargo\registry\src\github.com-1ecc6299db9ec823\panic-probe-0.3.0\src\lib.rs:72:9
   5: core::panicking::panic_fmt
        at /rustc/e1eaa2d5d4d1f5b7b89561a940718058d414e89c/library/core/src/panicking.rs:64:14
   6: core::panicking::panic
        at /rustc/e1eaa2d5d4d1f5b7b89561a940718058d414e89c/library/core/src/panicking.rs:114:5
   7: __defmt_default_panic
        at C:\Users\Mehmet Ali Anil\.cargo\registry\src\github.com-1ecc6299db9ec823\defmt-0.3.2\src\lib.rs:367:5
   8: defmt::export::panic
        at C:\Users\Mehmet Ali Anil\.cargo\registry\src\github.com-1ecc6299db9ec823\defmt-0.3.2\src\export\mod.rs:133:14
   9: qspi::____embassy_main_task::{{closure}}
        at src\bin\qspi.rs:77:13
  10: embassy_executor::raw::TaskStorage<F>::poll
        at C:\Users\Mehmet Ali Anil\Documents\embassy\embassy-executor\src\raw\mod.rs:161:15
  11: embassy_executor::raw::Executor::poll::{{closure}}
        at C:\Users\Mehmet Ali Anil\Documents\embassy\embassy-executor\src\raw\mod.rs:415:17
  12: embassy_executor::raw::run_queue::RunQueue::dequeue_all
        at C:\Users\Mehmet Ali Anil\Documents\embassy\embassy-executor\src\raw\run_queue.rs:70:13
  13: embassy_executor::raw::Executor::poll
        at C:\Users\Mehmet Ali Anil\Documents\embassy\embassy-executor\src\raw\mod.rs:395:13
  14: embassy_executor::arch::Executor::run
        at C:\Users\Mehmet Ali Anil\Documents\embassy\embassy-executor\src\arch\cortex_m.rs:55:17
  15: qspi::__cortex_m_rt_main
        at src\bin\qspi.rs:22:1
  16: main
        at src\bin\qspi.rs:22:1
  17: Reset
  (HOST) WARN  call stack was corrupted; unwinding could not be completed
  (HOST) ERROR the program panicked
error: process didn't exit successfully: `probe-run --chip nRF52840_xxAA target\thumbv7em-none-eabi\debug\qspi` (exit code: 22)

```

mehmetalianil, 2023-03-15:
This might be about #1262

mehmetalianil, 2023-03-15:
```rust
        unwrap!(q.blocking_erase(i * PAGE_SIZE as u32));
```
also fails similiarly

Dirbaio, 2023-03-15:
I just tried the example on a nrf52840-dk, it worked fine. What board are you using?

```
0.000000 INFO  id: [194, 40, 23]
0.000000 INFO  status: 66
0.000030 INFO  page 0: erasing... 
0.000640 INFO  programming...
0.124816 INFO  page 1: erasing... 
0.125427 INFO  programming...
0.242462 INFO  page 2: erasing... 
0.243072 INFO  programming...
0.367401 INFO  page 3: erasing... 
0.368041 INFO  programming...
0.492767 INFO  page 4: erasing... 
0.493377 INFO  programming...
0.618652 INFO  page 5: erasing... 
0.619262 INFO  programming...
0.738250 INFO  page 6: erasing... 
0.738891 INFO  programming...
0.858703 INFO  page 7: erasing... 
0.859313 INFO  programming...
0.979919 INFO  page 0: reading... 
0.980224 INFO  verifying...
0.981201 INFO  page 1: reading... 
0.981506 INFO  verifying...
0.982452 INFO  page 2: reading... 
0.982788 INFO  verifying...
0.983734 INFO  page 3: reading... 
0.984039 INFO  verifying...
0.985015 INFO  page 4: reading... 
0.985321 INFO  verifying...
0.986297 INFO  page 5: reading... 
0.986602 INFO  verifying...
0.987548 INFO  page 6: reading... 
0.987884 INFO  verifying...
0.988830 INFO  page 7: reading... 
0.989135 INFO  verifying...
0.990112 INFO  done!
0.990142 TRACE qspi: dropped
```

mehmetalianil, 2023-03-15:
I will take a look and try to reproduce. 
I have my own board but might also dig my nrf52840-dk.

mehmetalianil, 2023-03-17:
This was a configuration problem on my side. 
Need to use 32bit addressing.

closed 2023-03-17 by mehmetalianil