embassy-rs/embassy #1289: Changing baud on uart stm32

ra1u, 2023-03-20:
I would like to change uart baud at runtime.

Code seems to be in [`usart::configure`](https://github.com/embassy-rs/embassy/blob/b6663a013f8632cefbc6d6ab9a9aa7afbcd1314b/embassy-stm32/src/usart/mod.rs#L763-L803) and one needs to write appropriate value into [brr reg](https://github.com/embassy-rs/embassy/blob/b6663a013f8632cefbc6d6ab9a9aa7afbcd1314b/embassy-stm32/src/usart/mod.rs#L772)

However I can not figure out how to access relevant initialisation values [T::MULTIPLIER](https://github.com/embassy-rs/embassy/blob/b6663a013f8632cefbc6d6ab9a9aa7afbcd1314b/embassy-stm32/src/usart/mod.rs#L1110
) and T::frequency() because they are behind sealed (aka private) interface. Values are needed to calculate appropriate value for [`brr divisor`](https://github.com/embassy-rs/embassy/blob/b6663a013f8632cefbc6d6ab9a9aa7afbcd1314b/embassy-stm32/src/usart/mod.rs#L769) .

Question is, how do I change `brr` register, accordingly.
Only idea I have so far is to set value relative to current value. So if baud needs to go from 57600 to 115200 one can divide current  brr  reg by 2,

Is there anything better?

Dirbaio, 2023-03-20:
it's not implemented yet. The inner guts are intentionally not public, you're not supposed to touch registers directly if using the HAL. The fix would be to add a public `.set_baudrate()` method.

EnmanuelParache, 2024-11-06:
Just opened #3512 adding `.set_baudrate()` method as mentioned above.

Dirbaio, 2025-01-10:
done in https://github.com/embassy-rs/embassy/pull/3512

closed 2025-01-10 by Dirbaio