embassy-rs/embassy #1298: STM32H7 HAL: missing support for "wide VCO range"

diggit, 2023-03-25:
Hi,
STM32 H7 HAL does not seem to support VCO Wide Range mode. As a result, maximum frequency is capped at 420 MHz (H730 can do 550 MHz). Wide VCO range is 192 to 836 MHz.

Whole `pll` module would require some changes.
- VCO config: https://github.com/embassy-rs/embassy/blob/299689dfa2a5e160dbd6aa474772a9317a219084/embassy-stm32/src/rcc/h7.rs#L759
- asserts: https://github.com/embassy-rs/embassy/blob/299689dfa2a5e160dbd6aa474772a9317a219084/embassy-stm32/src/rcc/h7.rs#L715
- etc

I am total Rust noob (literally started 3 days ago), so I am not able to create PR with decent code quality.

I have one maybe a bit OT question, but would it be possible to make RCC config more compile time matter (optionally)?
At this moment all math is done in runtime and whole RCC config stuff takes over 1kiB on size optimized build.
In the end all checks could be done in compile time and then only registers would get filled (if config is compile time constant).
In C++, I'd leverage `constexpr` for this stuff, but I don't know options in rust and current HAL design.

diggit, 2023-03-25:
Also, PLL channel output P is always required even though, it is possible to disable it via config bit.
affected fcn: https://github.com/embassy-rs/embassy/blob/299689dfa2a5e160dbd6aa474772a9317a219084/embassy-stm32/src/rcc/h7.rs#L788
Should I open separate issue for this?

Dirbaio, 2023-03-26:
Please open PRs! Don't worry if you're new to Rust, we can work things out on PR review. :)

Dirbaio, 2025-01-10:
Fixed in embassy-stm32 v0.2.0

closed 2025-01-10 by Dirbaio