embassy-rs/embassy #1301: Embassy sync blocking mutex does not provide mutable reference

uzytkownik, 2023-03-27:
The purpose of mutex is to provide exclusive access. It seems that exclusive reference would be safe but currently it's not what interface provides:

```rust
    /// Creates a critical section and grants temporary access to the protected data.
    pub fn lock<U>(&self, f: impl FnOnce(&T) -> U) -> U {
        self.raw.lock(|| {
            let ptr = self.data.get() as *const T;
            let inner = unsafe { &*ptr };
            f(inner)
        })
    }

```

It would be better to have:

```rust
    /// Creates a critical section and grants temporary access to the protected data.
    pub fn lock<U>(&self, f: impl FnOnce(&mut T) -> U) -> U {
        self.raw.lock(|| {
            let ptr = self.data.get();
            let inner = unsafe { &mut *ptr };
            f(inner)
        })
    }
```

As we should have exclusive access to data.

lulf, 2023-03-27:
You can use a RefCell to get an exclusive mutable reference. See https://docs.rs/critical-section/latest/critical_section/struct.Mutex.html#design for more information.

Also see the existing issue https://github.com/embassy-rs/embassy/issues/1246

The documentation should probably be updated and more explicit about it.

uzytkownik, 2023-03-27:
I kind of see the argument though maybe there should be an `unsafe` version?

BTW - this is not consistent with async mutex where guard do implement `DerefMut`:

```rust
impl<'a, M, T> DerefMut for MutexGuard<'a, M, T>
where
    M: RawMutex,
    T: ?Sized,
{
    fn deref_mut(&mut self) -> &mut Self::Target {
        // Safety: the MutexGuard represents exclusive access to the contents
        // of the mutex, so it's OK to get it.
        unsafe { &mut *(self.mutex.inner.get()) }
    }
}
```

Sorry about #1246 but it didn't show in search.

uzytkownik, 2024-01-06:
Also there are Mutex implementations that should provide thread-safe access in embassy-rp.

closed 2025-01-10 by Dirbaio