embassy-rs/embassy #1304: BufferedUart split in embarry_rp returns tx/rx in reverse order than Uart

uzytkownik, 2023-03-28:
BufferedUart returns it in rx/tx order while Uart returns it in tx/rx oder:

```rust

    /// Split the Uart into a transmitter and receiver, which is particuarly
    /// useful when having two tasks correlating to transmitting and receiving.
    pub fn split(self) -> (UartTx<'d, T, M>, UartRx<'d, T, M>) {
        (self.tx, self.rx)
    }

    pub fn split(self) -> (BufferedUartRx<'d, T>, BufferedUartTx<'d, T>) {
        (self.rx, self.tx)
    }
```

It should probably be consistent.

wetheredge, 2024-08-13:
Looks like this was fixed in https://github.com/embassy-rs/embassy/commit/f733071908ced05bc006b43e1383336290faafdc

wetheredge, 2024-08-13:
[`BufferedUart::new`](https://github.com/embassy-rs/embassy/blob/f733071908ced05bc006b43e1383336290faafdc/embassy-rp/src/uart/buffered.rs#L93-L96) and [`BufferedUart::new_with_rtscts`](https://github.com/embassy-rs/embassy/blob/f733071908ced05bc006b43e1383336290faafdc/embassy-rp/src/uart/buffered.rs#L114-L119) still take interrupts before pins, while [`Uart::new`](https://github.com/embassy-rs/embassy/blob/f733071908ced05bc006b43e1383336290faafdc/embassy-rp/src/uart/mod.rs#L705-L708) and [`Uart::new_with_rtscts`](https://github.com/embassy-rs/embassy/blob/f733071908ced05bc006b43e1383336290faafdc/embassy-rp/src/uart/mod.rs#L729-L734) take pins then interrupts. Seems like pins then interrupts is more common in other peripherals.

JonathanPlasse, 2024-10-19:
Would a PR be welcome to take pins before interrupts on `BufferedUart`?

Dirbaio, 2024-10-21:
yes~

mgomez0, 2025-03-20:
Is this as trivial as it seems? I'm new to open source and the project in general. Also not sure what the scope of testing at my bench should look like, or if CI and inspection is enough here.

mgomez0, 2025-04-25:
@Dirbaio I think this can be closed now.

leftger, 2026-10-09:
This looks like it was addressed by f73307190, which changed "order of RX and TX handles in split fn, to streamline with other HALs".

`BufferedUart::split()` in embassy-rp now returns `(tx, rx)`, matching the other HALs.

Reopen or comment if you still see an ordering inconsistency somewhere.

closed 2026-10-09 by leftger