embassy-rs/embassy #1319: Issue with compiling for stm32f4

yashwanthsinghm, 2023-04-02:
I have been working on embassy but there compiling issue with the repository.

1)Tool chain log 

```
Default host: x86_64-apple-darwin
rustup home:  /Users/yashwanthsingh/.rustup

installed toolchains
--------------------

stable-x86_64-apple-darwin
nightly-2021-11-15-x86_64-apple-darwin
nightly-2022-09-18-x86_64-apple-darwin
nightly-2022-11-22-x86_64-apple-darwin
nightly-2023-02-07-x86_64-apple-darwin
nightly-x86_64-apple-darwin (default)
1.61-x86_64-apple-darwin

installed targets for active toolchain
--------------------------------------

thumbv6m-none-eabi
thumbv7em-none-eabi
thumbv7em-none-eabihf
thumbv7m-none-eabi
thumbv8m.main-none-eabihf
wasm32-unknown-unknown
x86_64-apple-darwin

active toolchain
----------------

nightly-2023-02-07-x86_64-apple-darwin (overridden by '/Users/yashwanthsingh/Yash/Projects/async/Embassy/git_embassyrust/embassy/rust-toolchain.toml')
rustc 1.69.0-nightly (e1eaa2d5d 2023-02-06)
```

2)error logs

```
yashwanthsingh@Yashwanths-MacBook-Pro stm32f4 % cargo build --bin blinky
    Updating crates.io index
  Downloaded stm32-metapac v2.0.0
  Downloaded 1 crate (5.1 MB) in 5.79s
   Compiling bitflags v1.3.2
   Compiling stable_deref_trait v1.2.0
   Compiling critical-section v1.1.1
   Compiling proc-macro2 v1.0.54
   Compiling proc-macro-error-attr v1.0.4
   Compiling proc-macro-error v1.0.4
   Compiling semver v1.0.17
   Compiling nb v1.1.0
   Compiling semver v0.9.0
   Compiling futures-core v0.3.27
   Compiling byteorder v1.4.3
   Compiling futures-task v0.3.27
error[E0463]: can't find crate for `core`
  |
  = note: the `thumbv7em-none-eabi` target may not be installed
  = help: consider downloading the target with `rustup target add thumbv7em-none-eabi`

error[E0463]: can't find crate for `compiler_builtins`

error[E0405]: cannot find trait `Clone` in this scope
   --> /Users/yashwanthsingh/.cargo/registry/src/github.com-1ecc6299db9ec823/stable_deref_trait-1.2.0/src/lib.rs:129:50
    |
129 | pub unsafe trait CloneStableDeref: StableDeref + Clone {}
    |                                                  ^^^^^ not found in this scope

error[E0463]: can't find crate for `core`
[... 905 lines cut ...]
   --> /Users/yashwanthsingh/.cargo/registry/src/github.com-1ecc6299db9ec823/byteorder-1.4.3/src/lib.rs:216:7
    |
216 |     + PartialEq
    |       ^^^^^^^^^ not found in this scope

error[E0405]: cannot find trait `PartialOrd` in this scope
   --> /Users/yashwanthsingh/.cargo/registry/src/github.com-1ecc6299db9ec823/byteorder-1.4.3/src/lib.rs:217:7
    |
217 |     + PartialOrd
    |       ^^^^^^^^^^ not found in this scope

error[E0405]: cannot find trait `Default` in this scope
    --> /Users/yashwanthsingh/.cargo/registry/src/github.com-1ecc6299db9ec823/byteorder-1.4.3/src/lib.rs:1809:6
     |
1809 | impl Default for BigEndian {
     |      ^^^^^^^ not found in this scope

error[E0405]: cannot find trait `Default` in this scope
    --> /Users/yashwanthsingh/.cargo/registry/src/github.com-1ecc6299db9ec823/byteorder-1.4.3/src/lib.rs:1839:6
     |
1839 | impl Default for LittleEndian {
     |      ^^^^^^^ not found in this scope

error: could not compile `byteorder` due to 89 previous errors
```

Dirbaio, 2023-04-02:
try running `rustup target add thumbv7em-none-eabi`. You have to do it on the exact nightly that Embassy uses, for example from some directory under the Embassy repo so that `rust-toolchain.toml` applies.

Though the rustup output you paste already indicates that's installed I think? If that's the case and it still doesn't work, perhaps something about rustup got corrupted, try removing `~/.rustup` and installing everything again.

yashwanthsinghm, 2023-04-09:
@Dirbaio i uninstalled and reinstalled rustup still the compilation error,

```
yashwanthsingh@Yashwanths-MacBook-Pro ~ % rustup show
Default host: x86_64-apple-darwin
rustup home:  /Users/yashwanthsingh/.rustup

installed toolchains
--------------------

stable-x86_64-apple-darwin
nightly-x86_64-apple-darwin (default)

installed targets for active toolchain
--------------------------------------

thumbv6m-none-eabi
thumbv7em-none-eabi
thumbv7em-none-eabihf
thumbv7m-none-eabi
thumbv8m.main-none-eabihf
wasm32-unknown-unknown
x86_64-apple-darwin

active toolchain
----------------

nightly-x86_64-apple-darwin (default)
rustc 1.70.0-nightly (af06dce64 2023-04-08)
```

Dirbaio, 2023-04-09:
yes, it should definitely work. There's something wrong with your Rust installation. Perhaps you have multiple of them, so it's uring the wrong one? Perhaps one through rustup, one not?

evanthomas, 2023-05-14:
I'm having a similar problem. Interestlingly, stm32f4 builds fine for me but nrf52840 is a mess. I've uninstalled and resintalled rust and cargo, WSL in my case.. Would love some help to debug this. 

```
evan@dugsy:~/projects/embassy/examples/nrf52840$ rustup show
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/evan/.rustup

installed toolchains
--------------------

stable-x86_64-unknown-linux-gnu (default)
nightly-2023-04-18-x86_64-unknown-linux-gnu

installed targets for active toolchain
--------------------------------------

riscv32imac-unknown-none-elf
thumbv6m-none-eabi
thumbv7em-none-eabi
thumbv7em-none-eabihf
thumbv7m-none-eabi
thumbv8m.main-none-eabihf
wasm32-unknown-unknown
x86_64-unknown-linux-gnu

active toolchain
----------------

nightly-2023-04-18-x86_64-unknown-linux-gnu (overridden by '/home/evan/projects/embassy/rust-toolchain.toml')
rustc 1.71.0-nightly (7908a1d65 2023-04-17)
```

```
evan@dugsy:~/projects/embassy/examples/nrf52840$ cargo run --bin blinky 2>&1 | head -40
warning: embassy-nrf52840-examples v0.1.0 (/home/evan/projects/embassy/examples/nrf52840) ignoring invalid dependency `probe-run` which is missing a lib target
   Compiling stable_deref_trait v1.2.0
   Compiling byteorder v1.4.3
   Compiling serde v1.0.163
   Compiling num-traits v0.2.15
error[E0463]: can't find crate for `std`
  |
  = note: the `thumbv7em-none-eabi` target may not support the standard library
  = note: `std` is required by `stable_deref_trait` because it does not declare `#![no_std]`
  = help: consider building the standard library from source with `cargo build -Zbuild-std`

error[E0463]: can't find crate for `std`
  |
  = note: the `thumbv7em-none-eabi` target may not support the standard library
  = note: `std` is required by `byteorder` because it does not declare `#![no_std]`
  = help: consider building the standard library from source with `cargo build -Zbuild-std`

error[E0463]: can't find crate for `std`
  --> /home/evan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/num-traits-0.2.15/src/lib.rs:21:1
   |
21 | extern crate std;
   | ^^^^^^^^^^^^^^^^^ can't find crate
   |
   = note: the `thumbv7em-none-eabi` target may not support the standard library
   = help: consider building the standard library from source with `cargo build -Zbuild-std`

error[E0463]: can't find crate for `std`
  |
  = note: the `thumbv7em-none-eabi` target may not support the standard library
  = note: `std` is required by `serde` because it does not declare `#![no_std]`
  = help: consider building the standard library from source with `cargo build -Zbuild-std`

error[E0432]: unresolved imports `self::core::cmp`, `self::core::iter`, `self::core::mem`, `self::core::num`, `self::core::ptr`, `self::core::slice`, `self::core::str`
   --> /home/evan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/serde-1.0.163/src/lib.rs:171:26
    |
171 |     pub use self::core::{cmp, iter, mem, num, ptr, slice, str};
    |                          ^^^  ^^^^  ^^^  ^^^  ^^^  ^^^^^  ^^^ no `str` in `lib::core`
    |                          |    |     |    |    |    |
    |                          |    |     |    |    |    no `slice` in `lib::core`
    |                          |    |     |    |    no `ptr` in `lib::core`
```

This results in ~5k errors.

evanthomas, 2023-05-17:
Try deleting and recloning the embassy repo, that worked for me. No idea how it got borked, all did was follow the instructions. Maybe the version I cloned was inconsistent.

yashwanthsinghm, 2023-05-17:
I reinstalled everything from scratch recloned the new repository, still the i'm getting same errors. @evanthomas @Dirbaio

Dirbaio, 2023-05-17:
make sure you've completely removed `/Users/yashwanthsingh/.rustup`, then let rustup redownload all the rust toolchain.

stappersg, 2023-08-13:
@yashwanthsinghm commented on May 17

> I was able to compile on my new linux machine now,
> i will remove .rustup and will check in mac.

Thanks for reporting _works on Linux_.
Please report _status on Mac_.
Closing the issue is also fine.

closed 2025-01-10 by Dirbaio