embassy-rs/embassy #1320: embassy-nrf - crates.io version is outdated?

dossalab, 2023-04-02:
Hi all,

I'm not really familiar with Rust infrastructure and tooling, so I'm sorry in advance if my question does not make sense.
I've been evaluating embassy for my embedded projects - so far I'm quite impressed with how many things are supported and how well this support looks like.

One thing that surprised me is that examples  are using relative paths when referring to the library code (I'm using [this one](https://github.com/embassy-rs/embassy/blob/master/docs/modules/ROOT/examples/basic/Cargo.toml) as an example):
```
embassy-executor = { version = "0.1.0", path = "../../../../../embassy-executor", features = ["defmt", "nightly", "integrated-timers"] }
embassy-time = { version = "0.1.0", path = "../../../../../embassy-time", features = ["defmt", "nightly"] }
embassy-nrf = { version = "0.1.0", path = "../../../../../embassy-nrf", features = ["defmt", "nrf52840", "time-driver-rtc1", "gpiote", "nightly"] }
```

Welp, I obviously don't want to use this approach, so I tried using versions from `crates.io`:

```
[package]
edition = "2021"
name = "embassy-oot-example"
version = "0.1.0"

[dependencies]
embassy-executor = "0.1.1"
embassy-nrf = "*"
embassy-time = "0.1.0"
```

That yields the following errors:
```
error[E0432]: unresolved import `embassy_nrf::gpio`
 --> src/main.rs:6:18
  |
6 | use embassy_nrf::gpio::{Level, Output, OutputDrive};
  |                  ^^^^ could not find `gpio` in `embassy_nrf`

error[E0433]: failed to resolve: could not find `main` in `embassy_executor`
 --> src/main.rs:9:21
  |
9 | #[embassy_executor::main]
  |                     ^^^^ could not find `main` in `embassy_executor`

error[E0425]: cannot find function `init` in crate `embassy_nrf`
  --> src/main.rs:11:26
   |
11 |     let p = embassy_nrf::init(Default::default());
   |                          ^^^^ not found in `embassy_nrf`
```

By the looks of it the version in `crates.io` was last updated around 2 years ago. So I assume the code there is not compatible with the modern version of executor.

I was thinking about using git version:
```
embassy-nrf = { git = "https://github.com/embassy-rs/embassy/" }
```
But this seem to search for `Cargo.toml` in the entire repository (which in this case contains a ton of sub-modules). I don't think there is a way to specify a concrete directory. So this just produces more errors ( however I'm not sure if those errors are related to other missing dependencies)

tommz9, 2023-04-03:
Hi. AFAIK the published crates are just placeholders until Embassy is more stable.

I use Embassy as a git submodule and then referring to it by a relative path in `Cargo.toml`:

```
my_repo/
  embassy/ (git submodule)
  my_fw/
    Cargo.toml (... path = "../embassy/embassy-executor" ...)
```

I like using it as a git submodule because it is easy to quickly update the whole repo when new changes are pushed upstream (just `git pull` on the submodule). I have it actually pointed into my own fork of Embassy since I made some changes to it.

dossalab, 2023-04-09:
That makes sense, I guess I'll try to implement it in a similar matter. Actually, I managed to use git in cargo like that:
```
[dependencies.embassy-nrf]
version = "*"
git = "https://github.com/embassy-rs/embassy"
features = ["defmt", "nrf52832", "time-driver-rtc1", "gpiote", "nightly"]
```
But I'm not sure how exactly that works, and how cargo picks the right subdirectory inside the common repo.

closed 2025-01-10 by Dirbaio